AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #257: Big Sur 11.2.3, app will not launch.

cspencer49519, 2021-04-22:
I receive the error
: "You can’t open the application “AlDente_2.2_Notarized” because it may be damaged or incomplete."

2018 MBP w/ touchbar.

MatthiasKerbl, 2021-04-22:
Hi @cspencer49519,

Did you have a previous version of AlDente installed?

cspencer49519, 2021-04-22:
I did, I removed it before I upgraded to Big Sur. I'm having problems with my battery and I'm hoping that I may have neglected to set the capacity back to 100%, as instructed.

cspencer49519, 2021-04-23:
What now happens is whenever my laptop is unplugged, once it reaches 70%, the screen turns off, and it logs me out, it won't turn back on until I plug it in. I have reset SMC and PRAM.

cspencer49519, 2021-04-23:
I should also add battery status reads normal under macos, and I've used my battery very minimally, so I'm just trying to make sure it's nothing I've done before I bring it to the apple store.

closed 2021-04-27 by cspencer49519