AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #260: AlDente Pro 1.0 Launch!

MatthiasKerbl, 2021-04-23:
After months of hard work, sweat and countless plates of pasta, we are proud to release the first public version of AlDente Pro.

Thanks to the incredible support of our beta testers, we can now bring AlDente Pro to everyone. You can get it at our new website: [https://apphousekitchen.com]

The pro version includes many convenient features like Sailing Mode, Heat Protection, Calibration Mode, Shortcuts and many more. On top of all that, it has a beautiful new design that gives you access to all important information at a glance.

If you would like to purchase our app and support further development of both the pro and the free version, head over to [AppHouseKitchen](https://apphousekitchen.com)!

Your’s truly,

David & Matthias

dstyp, 2021-04-23:
Nice, good work! Been following for a while, don't know how I missed the beta but just purchased.

First feedback: customizable menu bar icon to more minimal with less content. Got battery covered in istat menus :)

tosbsas, 2021-04-23:
awesome - so many thanks. Installed just perfect on top of beta. Anyway did restart my MBP afterwards

tosbsas, 2021-04-23:
what's a good setting for Temp - with the new sensor my MBP hardly loads :-))

MatthiasKerbl, 2021-04-24:
> what's a good setting for Temp - with the new sensor my MBP hardly loads :-))

Hi @tosbsas,
Apple recommends keeping your battery below 35 °C. I think that's a good starting point. If you want to be very cautious you can set it below 35 °C, maybe 32 °C or something like that. :)

char1eslu, 2021-04-24:
Will it be on Setapp？

Dhudong, 2021-04-24:
> After months of hard work, sweat and countless plates of pasta, we are proud to release the first public version of AlDente Pro.
>
> Thanks to the incredible support of our beta testers, we can now [...]

Hey Matt! Congrats on the launch of AlDente Pro. I've been waiting for this and I really wanna install it in my MacBook Air M1 but I was told that these types of applications may brick my mac or mess with my mac's battery. Is there any assurance that this app won't cause just that? Thank you.

MatthiasKerbl, 2021-04-24:
> Hey Matt! Congrats on the launch of AlDente Pro. I've been waiting for this and I really wanna install it in my MacBook Air M1 but I was told that these types of applications may brick my mac or [...]

Hi @Dhudong,

thank you very much. AlDente is installed and running on thousands of MacBooks for over a year now and no one reported a bricked device or any hardware issues. The only thing that some people reported was a disturbed battery calibration which can easily be avoided and even reversed by using our new feature Calibration Mode twice a month. Since we are using the same pathways as Apple to control the charging behavior of your MacBook, the risk of using AlDente is extremely low.

Kind regards,

Matthias

MatthiasKerbl, 2021-04-24:
> Will it be on Setapp？

Hi @char1eslu,

we are looking into this.

Kind regards,

Matthias

ModdedGamers, 2021-04-24:
What are the technical reasons behind AlDente Pro not being supported on Catalina or Mojave?

MatthiasKerbl, 2021-04-24:
> What are the technical reasons behind AlDente Pro not being supported on Catalina or Mojave?

Hi @ModdedGamers,

We used elements of SwiftUI which are only supported on Big Sur.
Are you still on Catalina or Mojave?

Kind regards,

Matthias

ModdedGamers, 2021-04-24:
I am stuck on Mojave currently.

Dhudong, 2021-04-25:
Hey David!

That’s good to know and thanks for the assurance! Anyway, I went ahead and installed the AlDente Pro yesterday and it’s working great and as intended.

Keep it up, man! We’re counting on your great app which no other app offers for mac.

Thanks!

Jaymar.

Dhudong, 2021-04-25:
> Hi @Dhudong,
>
> thank you very much. AlDente is installed and running on thousands of MacBooks for over a year now and no one reported a bricked device or any hardware issues. The only thing that some [...]

Hey David!

That’s good to know and thanks for the assurance! Anyway, I went ahead and installed the AlDente Pro yesterday and it’s working great and as intended.

Keep it up, man! We’re counting on your great app which no other app offers for mac.

Thanks!

dalisoft, 2021-05-02:
Thanks for such a great application.
Would be great if **Pro** version will be available on **Mac App Store** or **Homebrew** as Cask as for automatized scripts will be easier (e.g. Ansible playbooks)

closed 2021-07-04 by MatthiasKerbl