AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #261: macOS 10.13.6 support

jasper-zanjani, 2021-04-24:
Are there any plans to extend support to slightly older versions of macOS? I've been stuck on 10.13.6 for a long time because I'm running a 2011 early-model.

flatsiedatsie, 2021-06-06:
Similarly, I'd like to use this tool on MacOS 10.14.6 (Mojave). Later versions of MacOS have a battery charge limit built into the OS, so I was surprised to see this software only works on those later versions.

MatthiasKerbl, 2021-07-05:
Hi Everyone,
I have just included an answer to your question in our README:
"Due to SwiftUI not being compatible with older versions of macOS, AlDente is only compatible with macOS 11 Big Sur. Even though it might be possible to get AlDente to work on older MacBooks and macOS versions too, we are currently not working on it."

closed 2021-07-05 by MatthiasKerbl