AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #262: Multiple users logged in to MBP M1

AlTyndall, 2021-04-24:
I have noticed that I need to have AlDente running for all users who are logged in (I have fast user switching enabled), and the charge percentages all need to match. If I don't then the MBP seems to keep swapping between charging and not charging, probably as the various differently running user environments either limit to e.g. 75%, or 100% if AlDente isn't running in their user space. This is noticeable by the screen swapping back and forth from full brightness and dimmed as the MBP thinks it's running on battery and then becomes plugged in.

Would it be possible to set AlDente up so that it's set-and-forget in just one user environment (probably with Admin rights) and the effect will be applied across all users? That way, if I want to top up, I don't have to go into every logged-in user to do the same or log them out.

superluminal300, 2021-05-19:
+1 with this exact issue. The license with the associated settings should apply at the machine level and not the user level. Everytime someone else uses the laptop, Al Dente stops controlling the battery and the charge goes back up to 100%

MatthiasKerbl, 2021-07-05:
Hi Everyone,

thank you for reporting this issue. It's on our bug tracking list and we will try to do something about it.

evanvosberg, 2021-08-17:
My Intel MBP keeps swapping between charging and not charging. So the issue is not dedicated to M1 Macs.

Until now, I started AlDente just on one of the two logged in users. Will try to run it for both with same settings, but in general it should run on system level not for every single logged in user.

MatthiasKerbl, 2021-11-15:
Hi everyone,

We have just released a new version of AlDente (1.10). https://github.com/davidwernhart/AlDente/releases
This version has a fix for your issues with multiple users.

Thank you for your patience!

Cheers,
Matthias

closed 2021-11-15 by MatthiasKerbl

easz, 2024-08-21:
Hi @MatthiasKerbl,

just a follow-up question.

As stated on [fast-user-switching-explanation-and-tutorial](https://apphousekitchen.com/fast-user-switching-explanation-and-tutorial/) regarding to fast user switching, since version 1.15 we have to just install AlDente App for each user which can have different settings. Right? 

The text "When switching users or logging out, third-party apps like AlDente close" confuses me a little bit.
In case of "fast user switching" I can see multiple AlDente processes running for each user. (I assume it will still work as stated above) 

> usera@mac ~ % ps aux | grep -i alden
> usera        7664   0,4  0,6 409801648  50944   ??  S     9:20pm   0:41.74 /Applications/AlDente.app/Contents/MacOS/AlDente
> root              7481   0,3  0,1 408723744   7680   ??  Ss    9:09pm   0:21.33 /Library/PrivilegedHelperTools/com.apphousekitchen.aldente-pro.helper
> userb  7467   0,0  0,4 409940672  35776   ??  S     9:09pm   0:47.44 /Applications/AlDente.app/Contents/MacOS/AlDente
> usera       11160   0,0  0,0 408628368   1664 s000  S+    8:05am   0:00.00 grep -i alden
>

MatthiasKerbl, 2024-08-21:
> Hi @MatthiasKerbl,
> 
> just a follow-up question.
> 
> As stated on [fast-user-switching-explanation-and-tutorial](https://apphousekitchen.com/fast-user-switching-explanation-and-tutorial/) regarding to fast user switching, since version 1.15 we have to just install AlDente App for each user which can have different settings. Right?
> 
> The text "When switching users or logging out, third-party apps like AlDente close" confuses me a little bit. In case of "fast user switching" I can see multiple AlDente processes running for each user. (I assume it will still work as stated above)
> 
> > usera@mac ~ % ps aux | grep -i alden
> > usera        7664   0,4  0,6 409801648  50944   ??  S     9:20pm   0:41.74 /Applications/AlDente.app/Contents/MacOS/AlDente
> > root              7481   0,3  0,1 408723744   7680   ??  Ss    9:09pm   0:21.33 /Library/PrivilegedHelperTools/com.apphousekitchen.aldente-pro.helper
> > userb  7467   0,0  0,4 409940672  35776   ??  S     9:09pm   0:47.44 /Applications/AlDente.app/Contents/MacOS/AlDente
> > usera       11160   0,0  0,0 408628368   1664 s000  S+    8:05am   0:00.00 grep -i alden
> 
> AlDente Pro 1.27.3 macOS 14.6.1

Hi @easz,

Yes, AlDente will still work, you just need to have AlDente installed and running on all users you want to use it on.

Best,
Matthias