AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #263: Pro: Cannot disable the calibration mode toggle, if enabled in Clamshell mode

vividfog, 2021-04-24:
I've got an M1 Air running in clamshell mode. I clicked on the **Start Calibration** button.

**Observed behavior:** A dialog appears explaining that this mode can't be activated in clamshell mode. Clicking **OK** will make the dialog appear again, and again, and again. Calibration mode can't be turned off, even though it's not really on. Clicking **Quit** quickly after clicking **OK**, and then restarting the app, that's the workaround I used. Now the app works as usual.

**Expected behavior:** Clicking **OK** should turn Calibration Mode (button) off, and there should not be an infinite loop of dialogs explaining the situation.

[image]

MatthiasKerbl, 2021-04-24:
Hi @vividfog,

thank you for reporting this issue. We are looking into this.

Kind regards,

Matthias

MatthiasKerbl, 2021-04-25:
Hi @vividfog,

We have just released an update of AlDente Pro which fixed this issue.

Kind regards,

Matthias

closed 2021-04-25 by MatthiasKerbl

vividfog, 2021-04-25:
That was quick. I can verify that the issue is indeed gone after I updated the app.