AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #264: 12" MacBook (2015) and Al Dente Pro?

johnnyangel1, 2021-04-25:
Hello,

I have both an M1 MacBook Pro and a 12" MacBook (2015, with 1.1GHz Intel Core CPU) and have been using both Al Dente and Al Dente Pro on both. I've been delighted to have it (if Al Dente had existed earlier the 12" MacBook's battery would be so much better off!).

However, my experience has been that Al Dente Pro does not work on the 12" MacBook -- whereas the standard Al Dente _does_ work. And I noticed that the 12" MacBook is not listed on the AppHouseKitchen website as being a system that is supported.

Maybe I'm just doing something wrong? I've tried completely removing the app, the helper, etc. and reinstalling. It's not a big issue as I can just keep using the standard Al Dente on the old MacBook, but I'm just curious about what changed.

Cheers,
Jonathan

MatthiasKerbl, 2021-04-26:
Hi @johnnyangel1,

Unfortunately, due to technical limitations, the 12 inch MacBook is not supported by AlDente Pro. The only thing from AlDente Pro that should work on the 12" MacBook is the Charge Limiter when the Intel Mode is activated.

Kind regards,

Matthias

johnnyangel1, 2021-04-26:
Thanks, Matthias. Since the standard Al Dente does work nicely on the 12-inch MacBook, it’s not really a problem!

closed 2021-04-26 by johnnyangel1