AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #265: Pro download should not be behind trackers

gbil, 2021-04-26:
I paid for your app but the download link is behind trackers which I don't appreciate
specifically this part: https://click.pstmrk.it/ 

I was able to decode the tracking link from the email to the original one but it shouldn't be there to start with really

MatthiasKerbl, 2021-04-26:
> I paid for your app but the download link is behind trackers which I don't appreciate
> specifically this part: https://click.pstmrk.it/
> 
> I was able to decode the tracking link from the email to the original one but it shouldn't be there to start with really

Hi @gbil,

We use a third-party company to handle our worldwide transactions and it looks like they use are using this tracker.
Unfortunately, we can't do anything about it.

I hope you understand!

Kind regards,

Matthias

gbil, 2021-04-26:
not sure how this limits you from providing a direct link on your site

MatthiasKerbl, 2021-04-26:
> not sure how this limits you from providing a direct link on your site

You are right, I have just provided a direct download link in our FAQs on our website.

closed 2021-04-26 by MatthiasKerbl