AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #266: Pro version license key usage question

joshuarobs, 2021-04-27:
I've been looking forward to a pro version of AlDente as I've used this for many months with good results, so why not get a version that has more features and supports the devs?

My question is this: I only have 1 macbook right now, a 16" and I'm planning to eventually sell it and get a smaller 14" when it comes out. Your website says you can have up to 3 macbooks. Is that at a time? Or 3 total? I don't want to have to activate the license key on 2 macbooks, and only have 1 remaining upgrade left in the future, for the lifetime license. I had this problem with some other mac app where I had to buy a new license since I got a new laptop

Thanks, and keep up the great work!

driversti, 2021-05-07:
The answer to your question is there: [Pricing](https://apphousekitchen.com/pricing/)
**What Happens To My License When I Get A New MacBook?**
_No worries, you can take your license to your new MacBook. Just deactivate your license on the old MacBook and activate it on your new._

peros550, 2021-05-07:
> The answer to your question is there: [Pricing](https://apphousekitchen.com/pricing/)
> **What Happens To My License When I Get A New MacBook?**
> _No worries, you can take your license to your new MacBook. Just deactivate your license on the old MacBook and activate it on your new._

What if I have no other access to my old mac because i.e. hard drive crashed (happened to me recently). Is there a procedure to deactivate licenses remotely?

MatthiasKerbl, 2021-05-07:
> What if I have no other access to my old mac because i.e. hard drive crashed (happened to me recently). Is there a procedure to deactivate licenses remotely?

Hi @peros550,
yes, you would have to write to the support in this situation.

closed 2021-07-05 by MatthiasKerbl