AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #267: Not working on Big Sur 11.3

joshuarobs, 2021-04-27:
I just recently updated my mac to Big Sur 11.3 and realized that Aldente no longer works. The app runs and is in the task bar, and I set the limit to 50% but my battery is fully charged when connected

peros550, 2021-04-27:
For me (Apple MBP "13 2017 4ports)  seems to work

joshuarobs, 2021-04-27:
Yeah I think it was some weird bug on my machine somehow. It worked again when I clicked "Reinstall Helper".

closed 2021-04-27 by joshuarobs

pita13, 2021-06-09:
Hi there, I have a Macbook Pro 13" early 2015 and I can't get Aldente to work. After downloading and opening the app and installing the "helper tool" it says in the settings menu: Helper v12: not found! I tried destalling and reinstalling AlDente without success. Can somebody please help me. Kind regards, Peter