AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #269: Sailing mode doesn't work

Nemo3006, 2021-04-28:
Hey, I set my max chargelevel at 70%, sailing mode is activated with 5% but my MacBook Air doesn't sail. Is stays at 70% all the time that it is plugged in. What can I do, what can I test?

Greets, Daniel

MatthiasKerbl, 2021-04-28:
> Hey, I set my max chargelevel at 70%, sailing mode is activated with 5% but my MacBook Air doesn't sail. Is stays at 70% all the time that it is plugged in. What can I do, what can I test?
> 
> Greets, Daniel

Hey Daniel,

Check out our blog article about the Sailing Mode: https://apphousekitchen.com/feature-explanation-sailing-mode/

Kind regards,

Matthias

mikebore, 2021-05-01:
Thanks for the blog article. My experience is that the natural discharge is so slow that sailing does indeed appear not to be doing anything.

MatthiasKerbl, 2021-05-02:
> Thanks for the blog article. My experience is that the natural discharge is so slow that sailing does indeed appear not to be doing anything.

That's also my experience. However, you can speed up the Sailing by using an underpowered power supply.

closed 2021-05-02 by MatthiasKerbl