AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #271: After I uninstalled AlDente , Battery still charges.

YekeD, 2021-05-01:
I have 2020 MacBook Air M1, I uninstalled (It was set to %100) app and deleted helpers after that I restarted the pc but when I charge it ,even its %100 it says still charging. (On System Information) // Also when I plug the charging cable for a period of time it says "Battery is not charging" even though I deleted the app.

How can I solve this?

MatthiasKerbl, 2021-05-02:
Hi @YekeD,

I am not sure if I understand your issue correctly but you are saying that when you charge your MacBook to 100% your MacBook says that it is still charging even if the state of charge is already 100%? Secondly, when you are not at 100% and you have plugged in your MacBook, it does not charge?

YekeD, 2021-05-02:
Sorry for my English. The problem is after battery getting fully charged . And I  want to use Charging Adapter for laptop, On the Apple  menu and System Information/Power , it says charging, Basically overcharging.

YekeD, 2021-05-02:
"Secondly, when you are not at 100% and you have plugged in your MacBook, it does not charge?" I don't have problem with that. Just the overcharging problem.

YekeD, 2021-05-02:
[image]

> Hi @YekeD,
>
> I am not sure if I understand your issue correctly but you are saying that when you charge your MacBook to 100% your MacBook says that it is still charging even if the state of charge is [...]

[image]

MatthiasKerbl, 2021-05-02:
Hi @YekeD,

You don't need to worry about overcharging. Your MacBook has overcharging protection.
The reason why your battery still shows charging even if your state of charge is 100% is that it will rebalance your cells and calibrate your BMS.

closed 2021-05-02 by MatthiasKerbl