AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #272: Display doesn't work anymore

younes-ballout, 2021-05-02:
Hi, firstly sorry for the grammar errors english is not my first language.

I use my MacBook (M1) Pro connected to the Samsung M7 Monitor. 
The Laptop is connected with a USB-C (3.2 Gen2) Cable.
To protect my Battery from being charged the whole time to a 100% I installed AlDente.
Since then I was not being able to connect my Laptop to the Monitor anymore.
Everytime I put the USB C Cable inside the Laptop it starts charging for 1-2 Seconds and afterwards its stops charging.
Neither the MacBook or the Monitor does recognise each other anymore and I get no more screen playback. 
In an attempt to fix this problem i already reinstalled macOS Big Sur and installed all updates.
When I uninstalled AlDente I did put the charging limit to a 100% before.

Best regards, Younes

MatthiasKerbl, 2021-07-05:
Hi @younes-ballout,

do you still have this problem and are you sure it is because of AlDente?

ThatXliner, 2021-07-12:
> Everytime I put the USB C Cable inside the Laptop it starts charging for 1-2 Seconds and afterwards its stops charging.

I have uninstalled Aldente and this is happening

younes-ballout, 2021-07-12:
Hi @MatthiasKerbl,
to be honest i am not sure anymore. 
I read online a lot of stuff of similar complaints about this monitor.
I think this is a bug from the monitor (or macOS) and it had an unlucky timing while installing AlDente.
For myself i fixed the problem now using a thunderbolt 3 cable so i think its probably a really weird problem and it probably has nothing to do with AlDente.
Sorry for taking your time
Cheers

closed 2021-07-12 by younes-ballout

ThatXliner, 2021-07-12:
Coincidentally, I think I also found my reason to be unrelated

MatthiasKerbl, 2021-07-13:
> Hi @MatthiasKerbl,
> to be honest i am not sure anymore.
> I read online a lot of stuff of similar complaints about this monitor.
> I think this is a bug from the monitor (or macOS) and it had an unlucky [...]

Hi @younes-ballout,

I am glad you figured it out and it had nothing to do with AlDente.