AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #277: Mac doesn't charging without AlDente! Even after uninstalled

adayush, 2021-05-04:
My Mac (Air M1) doesn't charge without AlDente. I removed the app by setting the limit to 100% and then moved it to trash. It doesn't even charge now. Please help

adayush, 2021-05-04:
Nvm, after system update everything works nicely. Though the reboot didn't work when I tried it.

closed 2021-05-04 by adayush

ThatXliner, 2021-07-12:
I'm experiencing this issue, on an Intel Mac. Please help? I removed the app after uninstalling the helper, set "Optimize battery charging" back on, and moved it to trash.

MatthiasKerbl, 2021-07-13:
Hi @ThatXliner,

did you try an SMC reset? https://support.apple.com/en-us/HT201295

ThatXliner, 2021-07-13:
Sorry about this, this was probably caused by something else. Using another port for charging worked perfectly. It probably just got dirty.