AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #279: Mac M1: Not working Pro Version + High CPU usage

Kahmoon, 2021-05-07:
Hi guys,

i hope this is the right place for support. Yesterday i bought AlDente Pro. I followed the instructions for configuration but its not working. It say plugged + paused but it stays at 100%.

[image]
[image]
[image]

Also my system is lagging quite hard because this tool is requiring 98% of my cpu
[image]

I am on a mac m1
[image]

[aldentepro-debug-2021-05-07-09:25:26.pasta.zip](https://github.com/davidwernhart/AlDente/files/6439788/aldentepro-debug-2021-05-07-09.25.26.pasta.zip)

Im running AlDente Pro 1.01

Kahmoon, 2021-05-07:
Trying "using rosetta" workaround now

Also i updated to 1.03

davidwernhart, 2021-05-07:
Hey @Kahmoon 
Sorry for your inconveniences!

The new update should address some of the performance issues you are experiencing.
Also, AlDente Pro does not automatically discharge to the desired percentage if your state of charge is higher (100% in your case). You have to manually press the discharge button for this to happen.

Hope that helps,
David

Kahmoon, 2021-05-07:
Thanks for the fast reply. I try discharge.

I thought it would go down automatically until it reaches the 75% + sailing tollerance.

davidwernhart, 2021-05-07:
Hope that works for you!
We decided against automatically discharging to the desired percentage since it is needed very rarely and can confuse many users if always active.
Also, Sailing Mode does not actively discharge your device. Instead, it just waits until your battery discharges on its own (which can take a lot of time, mainly depending on your power brick) to charge it up again.
You can find some more information about it here: https://apphousekitchen.com/feature-explanation-sailing-mode/

Cheers,
David

Kahmoon, 2021-05-07:
I´m in discharge mode since almost one hour now and it still says 100%. Even the icon "plugged and discharge" is shown correctly. By the way: Charging source comes from my monitor via USB-C. 
[image]

Is there any way discharge work on clampshell mode? I m working 95% in clampshell mode on an external monitor? So its impossible to disconnect the charging power.

davidwernhart, 2021-05-07:
On M1 MacBooks, discharge can be very slow since the Apple Silicon chip really is sipping power.
You can try running a benchmark (or an 8k youtube video) to get your chip to sweat a little bit and consume more power.

Sadly, currently, we have not found a method to enable discharge in clamshell mode. Apple's software/hardware interface we are utilising does virtually disconnect all devices when enabled in clamshell mode. This means, that your monitor is getting disconnected when enabling discharge in this case, which kinda defeats the purpose.

Best regards,
David

Kahmoon, 2021-05-07:
OK

I have a strange behaviour now. First i unplugged my macbook from the power source the batterie icon showed "-" only. Since a few minutes it says "plugged and discharge". But im definately not plugged in right now.

[image]

After restarting the app it forces me to reinstall the helper tool. Icon is shown correctly after reinstall from helper tool

[image]

I´ll try 8k youtube video to get rid of some juice.

Kahmoon, 2021-05-07:
I think this tool is not working at all for me. I drained my Mac to 21%. Ich plugged in the power cable an it does not charge at all. Icon "plugged and paused".

[image]

When i press my mouse on the slider and hold down my left mouse it changes to charging for a split second.

[image]

After i completely kill/start the app it stays on charging. If i have to do this every time this is a bit unsatisfying :-)

davidwernhart, 2021-05-07:
Also, do you have any other apps active that might interact with the SMC? This includes most stats apps, MacsFanControl, or the free version of AlDente. Maybe one of them is interfering with AlDente Pro since both of them want to access the SMC at the same time.

Kahmoon, 2021-05-07:
Hmm nah, i m not running tools like this. Also this is my first try with AlDente. Never tried free version before.

Also its charging up to 100% without stopping at my set level of 75%. It should not go over 75% right?
[image]

I will give it a last try with your workaround of reinstalling helper tool tomorrow.

MatthiasKerbl, 2021-05-07:
Hi @Kahmoon,
Do you have Optimised battery charging disabled in system preferences?

Kahmoon, 2021-05-08:
Jepp, see the first screenshot in posted.

Kahmoon, 2021-05-10:
OK, i have to give up sorry. Every time i restart the tool it forces me to reinstall helper tool. Actually its charging to 100% again at did not stop at 75%.

I gonna contact you for a refund. sorry for the circumstances