AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #280: Feature request :Clamshell support for M1 Macs ?

F3000, 2021-05-07:
Is this something the Pro version can do ?

MatthiasKerbl, 2021-05-07:
> Is this something the Pro version can do ?

Hi @F3000,
AlDente Pro only has one feature which doesn't work in clamshell mode and that's Dicharge. This is due to technical limitations and we have not found a workaround for this issue yet. However, the other features such as Charge Limiter, Heat Protection and Sailing Mode work fine even in Calamshell mode.

F3000, 2021-05-07:
Thanks 😊 do you think that this could be added in a future update from Apple? 

And because of this the "calibration" Mode also don't Work in clamshell mode ?

MatthiasKerbl, 2021-05-08:
> Thanks 😊 do you think that this could be added in a future update from Apple?
> 
> And because of this the "calibration" Mode also don't Work in clamshell mode ?

It's unlikely but possible.
You are right, the Calibration Mode does not work in clamshell mode as well.

closed 2021-07-05 by MatthiasKerbl

F3000, 2021-08-19:
@MatthiasKerbl  

i have seen that with latest version 1.07 that issue would be solved ? 

So all features of the pro version including calibration mode are working in clamshell mode ?

MatthiasKerbl, 2021-08-22:
Hi @F3000,

A lot of development effort went into enabling these features but yes, Discharge and Calibration Mode work now even in Clamshell mode.

F3000, 2021-08-22:
@MatthiasKerbl thabks . Works for my M1 Air 😊

do you know something about the time how slow the system will lose  % ?