AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #282: Big Sur 11.3.1 seems to be invalid after the update. Is it just me. Every time you open it, you will be reminded to install the helper, but you are still charging

qiuniang6943, 2021-05-08:
Big Sur 11.3.1 seems to be invalid after the update. Is it just me. Every time you open it, you will be reminded to install the helper, but you are still charging
Model MBA M1
[image]

m12108, 2021-05-08:
I have the same problem

[image]

qiuniang6943, 2021-05-08:
> I have the same problem
> 
> [image]

My hint is the same as yours

ichbinsss267, 2021-06-07:
I am facing the same bug in 11.4 since in v. 2.2 since ... 2-3 weeks? That's too long, can't use AlDente anymore.

m12108, 2021-06-07:
> 我在 11.4 中面临同样的错误，因为在 v. 2.2 中，因为...... 2-3 周？太长了，不能再使用AlDente了。
My problem is solved
https://github.com/davidwernhart/AlDente/issues/283#issuecomment-841012993

ichbinsss267, 2021-06-08:
> #283 (comment) <https://github.com/davidwernhart/AlDente/issues/283#issuecomment-841012993>

Thank you, great! :)

MatthiasKerbl, 2021-07-05:
Hi everyone,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Please uninstall the old version of AlDente before installing the new one.

Sorry, it took so long but I hope you like the new version.

Kind regards,
Matthias

closed 2021-07-05 by MatthiasKerbl