AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #283: can't use. helper not found.i try 20times,and 2.1,its not work

ifan7, 2021-05-09:


eflylab, 2021-05-14:
[image]

+1
reinstall cant work

eflylab, 2021-05-14:
open terminal and run 
sudo launchctl load -w /Library/LaunchDaemons/com.davidwernhart.Helper.plist

MatthiasKerbl, 2021-05-14:
Hi @ifan7,

We are aware of this bug and working on a fix.
When an update is ready, we will let you know.

MatthiasKerbl, 2021-07-05:
Hi @ifan7 ,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Please uninstall the old version of AlDente before installing the new one.

Sorry, it took so long but I hope you like the new version.

Kind regards,
Matthias

closed 2021-07-05 by MatthiasKerbl