AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #290: Is it safe when the charge light blink orange to green?

dyaskur, 2021-05-15:
I just downloaded then restarted but still doesn't work, my MacBook Pro early 2015 still charging. Then I tried to check `Allow discharge` but it makes power source from battery and my charge light blinking every time

[image]

and also why still charging even above max battery charge setting?
[image]

dyaskur, 2021-05-15:
Sometimes after I plug out and plug in again the battery status full charge for abour 1-2 secon then charging again

MatthiasKerbl, 2021-07-05:
Hi @dyaskur,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases

Kind regards,
Matthias

closed 2021-07-05 by MatthiasKerbl

gahoo, 2023-07-12:
Version 1.22 has the same issue on MacBook Pro late 2013. In discharge mode, the led light keep blinking.

ext452, 2023-08-15:
I had the same issue when I connected the second monitor and used it for a while via type-c. After switching to HDMI I realized led blinking on power connector. Restarting of laptop helped me - no light blinking now.
MBP M1 14'
AIDente Pro 1.22.2