AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #294: Control charging with lid closed?

Peter376, 2021-05-25:
Hi @mzsombor86,

AlDente can not control the Charging behavior of your MacBook while it is in sleep. Therefore, it will keep charging until it is at 100%.

_Originally posted by @MatthiasKerbl in https://github.com/davidwernhart/AlDente/issues/252#issuecomment-821841381_

Peter376, 2021-05-25:
Hello all,
I came across this app and got very excited at first. I took a look at the forum pages and found [this topic](https://github.com/davidwernhart/AlDente/issues/252). If I understand correctly, AlDente cannot limit charging while the lid is closed. Is this correct?

Is this as designed? Our family has 4 M1 macbook Air units and yes the lid is closed often while charging.

Kind regards,
Pete

MatthiasKerbl, 2021-05-25:
> Hello all,
> I came across this app and got very excited at first. I took a look at the forum pages and found [this topic](https://github.com/davidwernhart/AlDente/issues/252). If I understand correctly, AlDente cannot limit charging while the lid is closed. Is this correct?
> 
> Is this as designed? Our family has 4 M1 macbook Air units and yes the lid is closed often while charging.
> 
> Kind regards,
> Pete
> 
> [Edit]
> I just wondered if the charge limit would work with lid open but machine left unused (in screen saver mode)

Hi @Peter376,

AlDente can only control charging, while your MacBook is turned on. However, our pro version has a feature called "Stop charging while sleeping" which will stop charging just before the MacBook goes into sleep. With this enabled, charging will be inhibited even if the lid is closed and the MacBook is still plugged in.

Kind regards,

Matthias

Peter376, 2021-05-28:
Dear Matthias,
Thank you for your info, I appreciate that. Though today I did a test and found that charging actually stops at the limit even with the lid closed. So now I am confused!

My test:
1. I set the battery charge limit in AlDente at 67%
2. I charged my old laptop (2013 MBP running Catalina and AlDente 2.2) to about 40%
3. I closed the lid
4. I waited a long time, more than enough to let it charge to 100%
5. I opened the lid and the battery was not fully charged, it was around the AD limit (69%)

So in fact this is the behaviour I always hoped to get so I'm very happy. But it doesn't seem to match what you stated above. Could you please clarify?

MatthiasKerbl, 2021-05-29:
Hi @Peter376,

In the comment above, I described the behavior of AlDente for M1 MacBooks.
On older MacBooks, like your 2013 model, the functionality is different. You can read more about the "Intel Mode" in our blog post: https://apphousekitchen.com/feature-explanation-intel-mode/

Your MacBook is asleep when you close the lid.

Kind regards,

Matthias

Peter376, 2021-05-29:
Thank you for the additional information. I apologise for the confusion. After reading the documentation / FAQ I thought Intel mode was only meant for bootcamp / Windows. But in the other thread you recommended to me to check this option for my macbook pro 2013 and that works very well, thanks for that. I now also understand that your comment only applies to M1 macbooks.

closed 2021-05-29 by MatthiasKerbl

vassudanagunta, 2021-05-30:
@MatthiasKerbl I'm a little confused. I have an M1 Macbook Air, but mine never charges when the lid is closed, plugged in, which is the state it is in most of the time when I'm not using it.  So either what you wrote above isn't quite right, OR my Al Dente is keeping my Mac running all the time just so it can prevent charging?

I'm running AlDente 2.2.

Peter376, 2021-05-30:
Do you have the Pro version installed? Or is it the free version?
The Pro version has a feature  "Stop charging while sleeping" which will stop charging just before the MacBook goes into sleep, as outlined above by Matthias.

vassudanagunta, 2021-05-30:
Yeah, I read that before I asked. It is because my Mac with free version doesn't charge with lid closed that I ask.

Peter376, 2021-05-30:
Okay sorry for that. Just thought I could provide quick troubleshoot just in case you didn't know about that Pro option. I hope support can help you further.

vassudanagunta, 2021-05-30:
no worries. And I have no complaint. I want it to behave just as it has been. But it differs from how @MatthiasKerbl *seems* to say it's supposed to work.

alvaroartano, 2023-10-04:
Hi. I have my Macbook Air M2. Al dente (free) keeps working even when  I close the lid. I have a 80% limit, and when I go to sleep, I wake up with it in a 80% battery level.

How is this happening if this is not the expected behavior?

Thanks