AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #295: Battery charge limit options for Late 2013 MacBook Pro 15"?

Peter376, 2021-05-25:
Hello, I downloaded AlDente (free version) and was planning on buying the pro version (as I usually tend to support developers), when I found that the unit keeps charging beyond the limit. After that I looked in the FAQ and found that pre-2016 models are not supported.

The unit is a Late 2013 MBP Retina 15". It's on Catalina 10.15.7. Is this a practical decision? Or is it impossible to access the SMC on older models?

Could there be another way / app to use on this machine?

Thank you.
Kind regards,
Pete

MatthiasKerbl, 2021-05-25:
Hi @Peter376,

AlDente works on your MacBook but you have to enable "Use Classic SMC key(Intel)".
Unfortunately, the pro features are not available for your MacBook model.

Kind regards,

Matthias

closed 2021-05-25 by MatthiasKerbl

Peter376, 2021-05-25:
Ok, thank you. I thought I read in the FAQ that that option was only for bootcamp setups. I'll try it.

chetanpm, 2021-10-08:
Set to 75%, it stops charging at 80. Thanks :)