AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #296: macOS 11.4 breaking AlDente Pro?

Swistunopolis, 2021-05-25:
Since yesterday's update to macOS Big Sur 11.4, AlDente is not working as expected; the icon is constantly changing between charging and pausing, but the battery is still getting charged.
I successfully switched over to the legacy Intel Mode (MacBook Pro 15" early 2019), which seemed to work, but after letting the unit sleep overnight I came back to a fully charged battery.

I've already tried reinstalling the helper tool, but that didn't solve the current behaviour.

MatthiasKerbl, 2021-05-25:
Hi @Swistunopolis,

AlDente Pro works perfectly fine on my Intel MacBook after the update.
Did you deactivate "Optimised battery charging" in System Preferences/Battery/Battery?

Swistunopolis, 2021-05-25:
Hi Matthias,

thanks for your response!
Yes, "Optimised battery charging" as well as "Manage battery longevity" had long been disabled; I just checked and they are still disabled.
Until the Update to 11.4, AlDente Pro had been working like a charm...

Swistunopolis, 2021-05-25:
I've now reinstalled AlDente Pro (which did not solve the problem) and gave the system another reboot - so far, it seems to work now!
As always, no one knows what exactly this reboot seems to have fixed.
If I don't report back here within 48 hours, consider this issue resolved. :)

MatthiasKerbl, 2021-05-25:
Have you tried to completely uninstall and reinstall AlDente Pro?

Swistunopolis, 2021-05-25:
You might have typed your response before noticing my recent post, haven't you? :)

MatthiasKerbl, 2021-05-26:
> You might have typed your response before noticing my recent post, haven't you? :)

You are right, I think we wrote the comments at the same time.
I am glad it works again. :)

itsoft3g, 2021-06-04:
For me too,  Pro works after restart(after the 11.4 update).

closed 2021-06-04 by MatthiasKerbl