AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #299: Crashes Mac

mygregstuff, 2021-05-25:
Crashes computer about 60 seconds after rebooting. SMC reset did not work.

Mac OS 11.3.1
MacBook Pro (15-inch, 2016)
2.9 GHz Quad-Core Intel Core i7

MatthiasKerbl, 2021-05-25:
> Crashes computer about 60 seconds after rebooting. SMC reset did not work.
> 
> Mac OS 11.3.1
> MacBook Pro (15-inch, 2016)
> 2.9 GHz Quad-Core Intel Core i7

Hi @mygregstuff,

Did you try to uninstall AlDente?

mygregstuff, 2021-05-26:
Yes. Aldente has been uninstalled.

MatthiasKerbl, 2021-07-05:
Do you still encounter this problem?
Did you delete the helpers too?

mygregstuff, 2021-07-06:
I have not re-encountered the issue after reinstalling the application.

MatthiasKerbl, 2021-07-06:
> I have not re-encountered the issue after reinstalling the application.

Perfect, you can try our newest update AlDente 1.04 which is much more stable than the old versions.

closed 2021-07-06 by MatthiasKerbl