AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #301: Unexpected heating M1 MBP

kolboch, 2021-05-28:
Hey, 
started using aldente few days ago and I've noticed that throttling charging comes with unexpected heating of the lower right part of the macbook (to the right of the trackpad). I think it's related to using AlDente, as pretty immediately after changing charging cap (eg. from 70 to 80) the heat soon is gone. I'm pretty worried about this one, as it would mean something is off with the software and it makes hardware work in unpredictable way. Not sure if the reason is aldente all the way I will keep you updated if I can confirm that 100%. My laptop usage is the same day2day, so my observations shouldn't be biased, but still I would keep an eye on what exactly is running at these times, observe heating up scenarios etc.
If you could look into that, or others will share their experience please do so.

Creeperface01, 2021-06-05:
In my case both of the bottom corners are constantly cold, and the main heat source are my hands

kolboch, 2021-06-06:
After further observations I think my concerns were irrelevant to AlDente.

closed 2021-06-06 by kolboch