AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #303: Aldente pro website blocked

sajeev2112, 2021-06-05:
Hi,
I would like to purchase AlDente but the website is currently blocked by my ISP (virgin media in the UK.) I would like to get the opportunity to get ardent pro so is there any alternative way where I can also reliably get updates.

hopkapi, 2021-06-06:
Have you tried [reporting it to VM](https://www.virginmedia.com/help/web-safe-troubleshooting)?

mikebore, 2021-06-07:
I am with Virgin Media in UK and site is not blocked for me.

What browser? If Safari, try click and hold the reload symbol in the Safari address window, and select "load without content blockers" from the drop down.

s4j33v, 2021-06-07:
I have tried it on different devices so I know it isn't a safari/chrome issue. It isn't a WebSafe issue or I would have seen the WebSafe pop up so I am not sure why this is an issue. I have realised this website works on a mobile data connection but I would like the website to work so I don't have any issues with future updates,

MatthiasKerbl, 2021-07-05:
> I have tried it on different devices so I know it isn't a safari/chrome issue. It isn't a WebSafe issue or I would have seen the WebSafe pop up so I am not sure why this is an issue. I have realised [...]

Do you think the problem might be your router/modem?

closed 2021-07-22 by MatthiasKerbl