AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #304: [M1] Resets to 0%

petrbela, 2021-06-07:
I was using AlDente on my old Intel MBP and it seemed to work as it should. Now I've migrated to an M1 Air (doing a 1:1 clone using Migration Assistant). When I try to set a percentage, it always resets back to 0 after a second.

Restarting nor uninstalling the app/helper didn't help, and as the M1 doesn't have a traditional SMC, I'm not sure what else I could check. FTR, before I did the migration, I briefly (for a few hours) tried AlDente on a fresh M1 system, and that seemed to work.

https://user-images.githubusercontent.com/668124/121040385-17d40400-c76f-11eb-893b-706b00e26d3a.mov

showsky, 2021-06-28:
Hi petrbela

you can disable rosetta and try it

[image]

petrbela, 2021-06-28:
That was a good pointer! I actually had to check "Open using Rosetta". Upon reopening, the helper was reinstalled, and the "Use classic SMC Key (Intel)" option showed up, which I unchecked, and now it's working as it should. Thanks!

[image]

closed 2021-06-28 by petrbela