AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #306: Works on mid-2015 retina MacBook Pro 15"

thejohnha, 2021-06-13:
Hi guys, just found your amazing software. Just wanted to let you know that it works perfectly on my mid-2015 retina macbook pro 15" running Big Sur. I saw that this model wasn't listed on your website, so I thought I'd let you know. Thank you again!

MatthiasKerbl, 2021-06-14:
Hi @thejohnha,

I am glad you like our app AlDente and thank you for reporting your MacBook model. I added it to supported devices.

closed 2021-06-14 by MatthiasKerbl

thejohnha, 2021-06-14:
@MatthiasKerbl the Intel mode works perfectly... Even when in sleep mode, it doesn't exceed the set threshold. Thanks again! I used to use the tape trick to block out the middle pin on the magsafe port, but this is so much better :-)