AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #307: Sailing Mode not working properly

Paco089, 2021-06-14:
Hello,

I use AlDente Pro for about 2 weeks now on my 16" MBP and the sailing mode does not work as intended. I have set my charge limit to 75% and sailing mode: Difference 35% ... will sail to: 40%. So in my understanding my MBP shouldn't charge back to 75% unless it hits 40%, but my machine already starts charging at 65%.

Is sailing mode not intended for such a wide range (35%) or what could be the problem?

Thank you in advance.

MatthiasKerbl, 2021-06-14:
Hi @Paco089,

We recommend using the Sailing Mode in a range of 5-10%. However, it should work for higher ranges as well but there are some events that can trigger a recharge to the upper limit, such as restarting or unplugging your MacBook.

Kind regards,
Matthias

Paco089, 2021-06-14:
Hi @MatthiasKerbl,

thank you for your quick response. Yes, I shutdown and unplug my machine every night and that's where the sailing mode is very helpful because sometimes it loses some % of battery overnight and if I plug it in again in the morning it won't charge. That also works 90% of the time. The problem seems to appear if I am outside that 5-10% range, meaning that if used my machine a few days like that (exclusively plugged in) and the loss of battery overnight sums up to > 10%, sailing mode won't work properly anymore.

I guess I will try to set it to "Difference 10% ...will sail to: 65%" and see if that works for better. Should be enough anyway because I rarely use my machine more than 3 days in a row exclusively plugged in and it won't lose more than 10% of battery in those 3 nights.

Kind regards

Paco

closed 2021-07-05 by MatthiasKerbl