AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #311: Top Up feature permanently resets the charge limit in 1.04

Dhudong, 2021-06-17:
Using Top Up feature ever since I've updated to 1.04 is a pain. It resets the charge limit to 100% everytime and I have to manually set it to previous limit for it to work again. Using M1 Macbook Air.

MatthiasKerbl, 2021-06-17:
> Using Top Up feature ever since I've updated to 1.04 is a pain. It resets the charge limit to 100% everytime and I have to manually set it to previous limit for it to work again. Using M1 Macbook Air.

Hi @Dhudong,

I think you found a bug in our latest version. We will definitely fix it with the next update.

Dhudong, 2021-06-18:
> Hi @Dhudong,
>
> I think you found a bug in our latest version. We will definitely fix it with the next update.

Please do, Matt. I use this feature quite often and it was working perfectly fine in previous versions.

MatthiasKerbl, 2021-08-25:
Hi @Dhudong,

This bug is gone with AlDente 1.07.

closed 2021-08-25 by MatthiasKerbl