AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #314: Overheating issue on Al Dente Pro 1.04

Dhudong, 2021-06-20:
[image]
My M1 Macbook Air has been overheating like crazy since I've updated to version 1.04 of Al Dente Pro. Just now it reached 43.3 degrees celsius. It normally sits at 34 degrees. I don't have any heavy app open and it's practically idle. I'm disabling the Al Dente Pro for now.

jacksoncooper, 2021-06-20:
Which system monitor are you using? I also have an M1 Air. I'd like to see if my machine is similarly affected.

Dhudong, 2021-06-22:
I'm using the heat protection feature of the Al Dente Pro which also shows the current temperature of the system.

vassudanagunta, 2021-06-22:
coconutBattery is a well regarded and popular tool for monitoring battery health. It includes temp.

MatthiasKerbl, 2021-06-26:
> [image]
> 
> My M1 Macbook Air has been overheating like crazy since I've updated to version 1.04 of Al Dente Pro. Just now it reached 43.3 degrees celsius. It normally sits at 34 degrees. I don't have any heavy app open and it's practically idle. I'm disabling the Al Dente Pro for now.

Hi @Dhudong,

depending on your ambient temperature, CPU load, and charging/discharging speed you battery can reach temperatures above 40 °C. That’s why we have implemented the feature Heat Protection. However, the CPU load of AlDente is negligible and therefore, AlDente does not really increase the temperature of your battery.

Do you still experience these high battery temperatures?

CaesireKin, 2021-07-01:
I have the same issues, the version on my mac book pro 2018 15-inch i7 is AI Dente Pro 1.04.
My mac book sleeps every night. in usual, it's cold when it wake up every morning. but since a few days ago, something get different, some sensors temperature goes to 40 ℃ above.
I've enable the sailing mode sail to 74%, and enable the Stop charging when sleeping, Stop charging when powered off. I think this is not caused by the ambient temperature, cuz I get up every day before dawn, it's significantly higher than the ambient temperature

MatthiasKerbl, 2021-07-05:
Hi @Dhudong @CaesireKin,

did you check the activity monitor if any process is using significant power and how much CPU AlDente Pro is using?

CaesireKin, 2021-07-08:
Hi @MatthiasKerbl ,
Thank you for your instructions, I would like to reset my mac book airflows setting to the system default (that is what I used when I found that problem) , and take some time to collect my mac book sensors log. 
btw, is AI Dente Pro SMC configs effect fans control?

MatthiasKerbl, 2021-07-11:
> Hi @MatthiasKerbl ,
> Thank you for your instructions, I would like to reset my mac book airflows setting to the system default (that is what I used when I found that problem) , and take some time to collect my mac book sensors log as well as the process cpu using.
> btw, is AI Dente Pro SMC configs effect fans control?

Hi,
AlDente Pro doesn't change the fan/airflow settings of your MacBook. Do you have another program which does?

udoersam, 2021-07-20:
If I hvae Al Dente Pro installed on my mac, I can recognize that the temperature is higher than without the app. It seems that the processor don't throttle down the speed to reduce the temp. If I use my macbook pro (Intel i7) in the clamshell mode the computer is crashing down when the temperature is to high.

MatthiasKerbl, 2021-07-23:
> If I hvae Al Dente Pro installed on my mac, I can recognize that the temperature is higher than without the app. It seems that the processor don't throttle down the speed to reduce the temp. If I [...]

Hi @udoersam,

Which MacBook model do you have?

CaesireKin, 2021-07-23:
Hi @MatthiasKerbl 

I think I was posted a problem caused by my own. follow your instructions, I've checked my mac sensors log and cpu usage log, it's seems the temperature growing up every day at 6:00 AM, that is the notification time of my mac's calendar schedule, it wake up the mac book, and seems take a lot of time to run some other program.
Sorry about the issues, maybe it's a problem about system optimization, not caused by AI Dente Pro

MatthiasKerbl, 2021-07-23:
> Hi @MatthiasKerbl
>
> I think I was posted a problem caused by my own. follow your instructions, I've checked my mac sensors log and cpu usage log, it's seems the temperature growing up every day at [...]

Hi @CaesireKin,

Thanks for the update and I am glad that AlDente Pro is most likely not the reason for your heat issues.

closed 2021-08-25 by MatthiasKerbl