AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #315: Not working. Continues to charge to 100%

dmelgar, 2021-06-20:
### Discussed in https://github.com/davidwernhart/AlDente/discussions/313

<div type='discussions-op-text'>

<sup>Originally posted by **dmelgar** June 19, 2021</sup>
I have a MacBook Air M1.
I am running MacOS Monterey 12.0 Beta (21A5248p) - ie beta 1.
I've installed AlDente.
AlDente is running. It appears in the menu bar.
I have it set to 75%
It has (had) been working.
I plugged in the MacBook Air while it was closed, ie clamshelled. 
When I opened the MacBook Air it was at 100%.

Is there some limitation regarding when it works? Does it only work if the MacBook isn't closed?</div>

Brewforce, 2021-06-22:
Same behavior here with a MacBook Air M1 running Big Sur 11.4.

Brewforce, 2021-06-23:
Answer from AlDente Team  if by "closed" you mean put to sleep, this may help you : 

"AlDente cannot operate in sleep mode, since macOS does not allocate CPU time to 3rd party apps when sleeping.
Because of that, AlDente is not able to actively monitor your battery status and switch off charging at the right time when you close the MacBook.
The only thing we are able to do is detect when the system is about to go to sleep and stop charging at the current percentage beforehand. This is what the "stop charging when sleeping" toggle does.

Hope that clears things up,
David"

dmelgar, 2021-06-23:
@Brewforce I hadn't seen that from the "AlDente Team". Is that in a FAQ somewhere or in the readme? Is there a link?
Aldente DOES appear to work even when the laptop is asleep for Intel Macs. But it is not behaving that way for M1 Mac. Apparently the difference is the SMC value stored for Intel changes the behavior of the firmware, whereas it doesn't work on M1 and Aldente has to actively manage battery charging.

I also do not see a "stop charging when sleeping" toggle. Where is that?

Brewforce, 2021-06-24:
Is an answer they give me on a Pro version discussion but the option seems to be only in the Pro version of the software.

closed 2021-07-21 by MatthiasKerbl

flow0103, 2022-07-01:
Okay, for me it does not stop charging in general. Regardless of sleep or active. Any idea as to why? Set limiter but still happily charges away.
MAC Batterie optimizer is turned off. Anything else I should check?