AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #320: Al Dente 2.2 + Apple's Optimised battery charging together may be better? M1 2020 Air

Peter376, 2021-06-24:
Hey all,
I read the FAQ, it tells me to disable Apple's Optimised battery charging as to prevent any interference with Al Dente. I understand that. However, with lid closed Al Dente cannot be activated and charging will proceed to 100%.

So this is what I was thinking:
- I have 3 kids using an M1 2020 MBAir and they couldn't care less about cooking their laptops batteries
- They typically look at the charge level after finishing their work, connect the charger and push the machine aside
- Therefore it will always charge to 100%, even with Al Dente enabled
Solution (or at least improvement):
- Set Al Dente to stop charging at 70%
- enable Apple's Optimised battery charging, which will use 80% as the threshold
Scenario 1: charging with lid open :
- Al Dente will prevent charging beyond 70%, thus not interfering with Apple's Optimised battery charging (80%)
- Even if charging goes a little over 70%, it will stay below 80%
Scenario 2: charging with lid closed:
- Al Dente will not be actively monitor battery level
- Apple's Optimised battery charging will prevent charging over 80% _for a while_. I am aware it will eventually charge on to 100%, but there is a period when it will stay at 80%
- If during that period the user opens the lid (kids do that now and then to check their school stuff), Al Dente will detect charge is over 70% and will become active, so charging will stop
- In this scenario, Al Dente may interfere with Apple's Optimised battery charging. Is this a bad thing?

So by enabling Apple's Optimised battery charging feature, we can prevent numerous 100% charge cycles, given our use case.

Please advise if the interaction between Al Dente and Apple's Optimised battery charging can lead to serious mishap and should be avoided at all cost.

Note:
On one of the kids M1 units I installed Al Dente (non Pro) and forgot to uncheck Apple's Optimised battery charging. I wrote a script that writes all battery charge/release cycles to a log file and noticed that it hardly ever reached 100%. The script keeps track of battery load percentage, charger status (connected or disconnected), lid open/closed and time.

Thanks,
Peter

dalisoft, 2021-09-01:
I using this as this way

- Enabled Apple Optimized battery mode
- Set max to 70% on Al Dente (anyway goes up-to 80%)
- Enabled sailing mode and set to 35% unit (from 70% max back to 35% then charges back to 78-80%)

Works fine. Every week (Sunday) i calibrate battery to return back battery capacity and check via **CoconutBattery**.

asher-gh, 2021-11-30:
> Weeks before, I wrote a script that writes all battery charge/release cycles to a log file and when reviewing those logs I noticed that that machine _hardly ever reached 100%_. The script keeps track of battery load percentage, charger status (connected or disconnected), and time. As it runs every 10 minutes, prolonged intervals indicate when the lid was closed. This script also sends notifications to the user to urge them to stop or start charging when required.

could you share the script? Would love to have a notification to remind me to charge at set percentage

Peter376, 2021-11-30:
Certainly. It's been a while so I'll need to dig up the source code. Also, I may need to clean up some test code and maybe add some explanation of how it all works. Very busy working weeks until xmas so I don't know when I will be able to do it. But I will be back here 100%.

Peter376, 2021-12-07:
Hi Asher,
It turns out I created several scripts. The original one sends notifications and creates a log file. MacOS only. Then I have two more that only log battery levels at the extreme ends. 
The original script grew quite big, very extensive now with lots of user variables and multiple levels of alerts (alerts at 35%, 30%, 25%, 20%, 10% and 5%, also at 70%, 75% and 80%). This may discourage you a bit. 
I could slim down the original script so it will only give you one alert at each end.
Are you comfortable running scripts and creating plist for automated execution?
Pete

asher-gh, 2021-12-07:
Oh wow, I guess I would be able to do it. But would appreciate some pointers. Thanks for your time and effort

Peter376, 2021-12-10:
I cleaned up my script. I can send it to you if you like. I used another script from github user Jessica Moloney: https://github.com/jmmoloney/batteryscript. If you find mine too extensive, you may want to have a look at the original.
Please read the comment sections carefully. You may want to read Jessica's instructions as a starting point, but beware that simply exporting the script from script editor "as script" won't work on M1 Macs. Please use the manual walkthrough in Terminal as explained in the comments.
If you have any questions, please ask.
Pete
[batteryScript 2_0 kopie.zip](https://github.com/davidwernhart/AlDente/files/7692182/batteryScript.2_0.kopie.zip)

Peter376, 2021-12-10:
FYI: [This](https://apple.stackexchange.com/questions/413619/sound-often-randomly-disappears-from-menu-bar/413629#413629) is the source I used for the manual workflow in Terminal.

Peter376, 2021-12-11:
Just another FYI I am currently doing a complete overhaul. Your interest in the script sparked a total cleanup but it is still too comprehensive for the average user. I don't want to create different forks so I am trying to make the script more basic by default and move the multiple alert options to a sub routine. I have also created a config file so we don't need to compile everything again after changing message limits.

Peter376, 2021-12-19:
@asher-gh Have you been put off much by my previous batteryScript version or are you in for a new release?

asher-gh, 2021-12-19:
Thank you so much. Honestly I’m pretty satisfied with the stripped down version to just have a reminder to charge and I usually set a timer and remove the charge after 15-20mins.

Peter376, 2022-01-24:
If it's only the charge reminder you need then you can check out coconutbattery. 
Also: how much charge does your charger add to  your battery in just 15-20 mins? Our macbooks M1 go up 1% a minute.

StevenCurran, 2022-02-02:
Using al dente free -> along with optimized charging I am not seeing any battery discharge / recharge when my m1 air is closed, and with the charging cable plugged in. Essentially al dente is just avoiding the overcharging past 70% for me. Is there something that has been added for m1 macs with the built in optimized charging at the os level that is doing this?

closed 2023-11-15 by MatthiasKerbl