AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #322: Our M1 MacBook Air units don't go to 100% even with lid closed

Peter376, 2021-06-26:
Let me show you the following log file, which is populated by an Applescript routine.
I created a plist in ~/Library/LaunchAgents/ which runs the Applescript (made Executable) every 1 minute.

Every minute, the log is refreshed with the current battery charge, the time and - when charging - Wall Power.

The test below is with the following settings:
- M1 Macbook air 2020 16GB
- Al Dente 2.2 (free version) downloaded last week
- Al Dente max. battery charge set to 85% for this test, but also works the same at 70%
- M1 Optimised battery charging disabled
- Lid closed and Wall Power connected as from 22:27

As you can see, even with the lid closed, the battery will not charge beyond the % set in AD.

In [issue 256](https://github.com/davidwernhart/AlDente/issues/252) (closed), it is clearly stated that:

> Hi @mzsombor86,
> AlDente can not control the Charging behavior of your MacBook while it is in sleep. Therefore, it will keep charging until it is at 100%.

At first, I was a bit disappointed about this limitation of setting a battery charge delimiter such as Al Dente. But in my testing I found that it is now behaving the way i would like it to be.

But why is my M1 behaving differently? I am posting this because if this turns out to be a new insight, it could contribute to the development of Al Dente.

Here is the log:

71 - 22:23:06 - Wall Power
71 - 22:24:06
71 - 22:25:07
71 - 22:26:07
71 - 22:27:08
71 - 22:28:08 - Wall Power
71 - 22:29:09 - Wall Power
71 - 22:30:09 - Wall Power
73 - 22:31:09 - Wall Power
73 - 22:32:10 - Wall Power
74 - 22:33:10 - Wall Power
75 - 22:34:11 - Wall Power
75 - 22:35:11 - Wall Power
76 - 22:36:12 - Wall Power
77 - 22:37:12 - Wall Power
77 - 22:38:13 - Wall Power
78 - 22:39:13 - Wall Power
78 - 22:40:14 - Wall Power
79 - 22:41:14 - Wall Power
80 - 22:42:15 - Wall Power
80 - 22:43:15 - Wall Power
81 - 22:44:16 - Wall Power
82 - 22:45:16 - Wall Power
82 - 22:46:16 - Wall Power
83 - 22:47:17 - Wall Power
83 - 22:48:17 - Wall Power
85 - 22:49:29 - Wall Power
85 - 22:55:03 - Wall Power
85 - 22:56:12 - Wall Power
85 - 23:11:19 - Wall Power
85 - 23:12:20 - Wall Power
85 - 23:37:40 - Wall Power
85 - 23:54:21 - Wall Power

Pete

Peter376, 2021-06-26:
Here are the energy settings of the unit I used in this test:

[image]
[image]
[image]
[image]

MatthiasKerbl, 2022-03-03:
Hi @Peter376,

Unfortunately, we could not get AlDente to work reliably in sleep mode due to third-party apps not being able to run when the MacBook is asleep.
That is why we introduced the features "Stop charging while sleeping" and "Disable sleep until charge limit".
You can read more about those features on our blog:

https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl

Peter376, 2022-03-03:
Dear @MatthiasKerbl 
Thank you for your kind reply and for your time looking into this matter. The point I was trying to make, though maybe a tad inarticulately, is that AD - despite not being designed or even being considered unable to operate with lid down - actually _does_ keep our M1 macbooks safe below the boundary set by the user. Free version used for the test.

I am intrigued as to why this could be.

Kind regards,
Peter

MatthiasKerbl, 2022-03-04:
One theory I have is that the new Apple Silicon MacBooks have the power nap feature activated by default and probably wake up many times during the sleep. When they wake up, AlDente most likely also gets some cpu time and can prohibit charging when the SOC is at the charge limit and when charging is prohibited while the MacBook goes into sleep, it will not continue charging. However, we did not get a too consistent result especially considering older MacBook models. That is why we took a different path with our features but we will keep our eyes open and when we find an improvement, we will definitely update AlDente.

Cheers,
Matthias