AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #323: Feature Request - Low battery notification

Jseeca, 2021-06-27:
From what I'm aware I don't think Mac's have an inbuilt low battery notification. I think it would make a lot of sense to add an option for AlDente to notify you at a specified percentage, say 25%. Also for the top up function, I think it would also be good to give users control over the topup percentage, not just default 100%. 

There are certain cases like me were I'm at home for most of the week and it make more sense for me to have the limit at 50% since it's plugged in all the time, and the days I do need to leave the house, I don't actually need 100% anyways, so I could "top up" to 80%. 95% of the time I don't need 100% charge because the battery on the M1 is so good.

MatthiasKerbl, 2021-06-28:
Hi @Jseeca,

both of your suggestions sound good. I will put them on our list for future updates.

cyriltvarghese, 2021-07-02:
Hello Matthias and David!

When can we expect Al Dente for Monterey ? 
I'm on version MacOS 12.0 Beta (21A5268h) and Al Dente does ( Free ) is not working. 
Thanks for making this app!