AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #330: Charge On Hold icon no longer working properly

LeshawnG, 2021-07-05:
I uninstalled and reinstalled AlDente today. and upon reinstalling, it no longer displays the charge on hold icon with the plug over the battery, like this.
[image]

When I plug in the charger now, it shows the battery always full at 100 and not the %charge it is held at like it did before.

[image]

I know this is minor but I really prefer seeing the charge on hold icon. Is there anyway to fix this?

Also, how does AlDente display for you? Does it show 100% constantly when plugged in or does it show the percent with the charge on hold icon?

MatthiasKerbl, 2021-07-05:
Hi @LeshawnG,

do you have AlDente Pro installed?

LeshawnG, 2021-07-05:
@MatthiasKerbl 
I don't mean to create a false alarm but the "charge on hold icon" is back to working correctly now on AlDente Classic 2.2.

The issue first occurred upon installation of AlDente 1.04 and still persisted in AlDente Classic 2.2 upon reinstallation (1.04 was uninstalled prior to 2.2 reinstallation).
I also tried running older versions of Classic 2.0 and 2.1.2 where the issue persisted before reinstalling 2.2 again.

The issue persisted on 2.2 for a few hours yesterday after final reinstallation while my computer was in use. Today the icon is working normally.

I am not sure what caused the issue upon installation of 1.04 and I can't say what fixed it.

If anything persists in the future I will be sure to let you know. 
Thank you for the quick response!

closed 2021-07-05 by LeshawnG