AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #331: Battery max limit

tymoey, 2021-07-09:
Hello, my MBA max battery is now at 88%, I've tried to uninstall Aldente followed by the reboot but the charging still does not goes over 88%, let me know if there is anything can be done?

MatthiasKerbl, 2021-07-13:
Hi @tymoey,
have you tried the following:

1. Turn off your MacBook
2. Close the lid for at least a minute
3. Turn it on again and let it charge

Did it go to 100% again?

eianzhang, 2021-07-13:
I have the same problem. I set it as 95, then I won't be able to charge it to 100. I pull back the scroller back to the rightest point at 100, waited for 10 minutes for it took effect, uninstalled, rebooted, still, I can't make it back. Finally, I reinstalled Big Sur 11.4, but it didn't work, either.

eianzhang, 2021-07-13:
> Hi @tymoey,
> have you tried the following:
> 
> 1. Turn off your MacBook
> 2. Close the lid for at least a minute
> 3. Turn it on again and let it charge
> 
> Did it go to 100% again?

for me ,it didn't work

markwu, 2021-07-14:
Me either. It doesn't work at all.

markwu, 2021-07-14:
After install/uninstall over 10 times, finally, it works. No idea what's going on.

Resserene, 2021-07-16:
I am having the same issue. Installed/uninstalled several times. Reset the SMC more times than I can count. Even factory reset  my MacBook. Mine is somehow stuck at 1%. So its not charging the battery at all. I might have to buy a new computer if this persists. Couldn't imagine an app could mess up my system so bad.

markwu, 2021-07-19:
> I am having the same issue. Installed/uninstalled several times. Reset the SMC more times than I can count. Even factory reset my MacBook. Mine is somehow stuck at 1%. So its not charging the [...]

I think the developer should provide a resetting tool to prevent this.

MatthiasKerbl, 2021-07-20:
Hi @tymoey, @eianzhang, @markwu, and @Resserene,

I am sorry you encountered a new potential bug of AlDente.

Please answer the following questions:

1. Which MacBook model do you use?
2. Which macOS version do you have installed?
3. Which version of AlDente do you use?

Please generate and share a debug file for more information.

MatthiasKerbl, 2021-07-20:
> I think the developer should provide a resetting tool to prevent this.

With the Remove button in the settings of AlDente, you can reset every Setting AlDente changed and remove all of the Helper files.

eianzhang, 2021-07-21:
> Hi @tymoey, @eianzhang, @markwu, and @Resserene,
>
> I am sorry you encountered a new potential bug of AlDente.
>
> Please answer the following questions:
>
> 1. Which MacBook model do you use?
> 2. Which [...]

1. Macbook pro 13, 2017
2. Big Sur 11.4
3. AIDente 1.04

Resserene, 2021-07-21:
> Hi @tymoey, @eianzhang, @markwu, and @Resserene,
>
> I am sorry you encountered a new potential bug of AlDente.
>
> Please answer the following questions:
>
> 1. Which MacBook model do you use?
> 2. Which [...]

1. MacBook Pro late 2013
2. MacOS 11.4
3. AlDente 2.2

I don't see a Remove button in my AlDente settings. There is only a Reinstall Helper button.

markwu, 2021-07-21:
> Which MacBook model do you use?

Macbook Pro Retina 15" 2013

> Which macOS version do you have installed?

I install AlDente in macOS Catalina, and remove it. But the battery max limit stay 70%. Then, I try to upgrade my macOS to Big Sur, it still stay with 70%.

After install/uninstall over 10 times, and play with install `helper` and use "SMC classic" settings, finally the mac battery limit back to 100%. No idea what's going.

> Which version of AlDente do you use?

The lastest one downloaded from github. It should be `AlDente 1.04`.

> With the Remove button in the settings of AlDente, you can reset every Setting AlDente changed and remove all of the Helper files.

There is no `Remove` button.

NeilEvans22, 2021-07-22:
I also have this issue (new user as of yesterday)
Charge is stuck at 84% (check "About his Mac" and under Power it says "not charging"
Using ALDente 1.04 (new/virgin install)
Have a MacBook Pro mid-2014 15" using Bug Sur 11.4
Have de-installed ALDente using the REMOVE button as well as via an Uninstaller to get rid of everything
Shutdown and restarted ... still won't change past 84%
Pretty disappointed - do you test this stuff before you release it? ... 
Yes it's a free version and I guess you get what you pay for - won't be using this again :(
Please fix ASAP .. I want my Mac working again!!

NeilEvans22, 2021-07-22:
Couldn't wait.
I re-installed v1.04 again
Moved the slider to the right (100%)
Mac is now charging ... this code is scary ... so I'm gong to remove it completely .... good luck everyone

MatthiasKerbl, 2021-07-22:
> Macbook Pro Retina 15" 2013
>
>
> I install AlDente in macOS Catalina, and remove it. But the battery max limit stay 70%. Then, I try to upgrade my macOS to Big Sur, it still stay with 70%.
>
> After [...]

Hi @markwu,

AlDente sets a charge limit for older MacBooks which will stay active even if you close or deinstall AlDente. To reset the charge limit to 100% again you have to change it to 100% again with AlDente or do an SMC reset of your MacBook.

We know this behavior of AlDente is not ideal and we will implement multiple safety measures in an update to avoid this from happening again.

AlDente 1.04 should work perfectly fine on your MacBook. Just keep in mind to reset the charge limit to 100% before closing/uninstalling the app until the new update comes out.

You can find the Remove button in Settings->General Settings->Remove.

Kind regards,

Matthias

MatthiasKerbl, 2021-07-22:
> Couldn't wait.
> I re-installed v1.04 again
> Moved the slider to the right (100%)
> Mac is now charging ... this code is scary ... so I'm gong to remove it completely .... good luck everyone

Hi @NeilEvans22,

Thank you for your detailed description.

AlDente sets a charge limit for older MacBooks which will stay active even if you close or deinstall AlDente. To reset the charge limit to 100% again you have to change it to 100% again with AlDente or do an SMC reset of your MacBook.

We know this behavior of AlDente is not ideal and we will implement multiple safety measures in an update to avoid this from happening again.

AlDente 1.04 should work perfectly fine on your MacBook (as it does for thousands of other users). Just keep in mind to reset the charge limit to 100% before closing/uninstalling the app until the new update comes out.

Kind regards,

Matthias

MatthiasKerbl, 2021-07-22:
> 1. Macbook pro 13, 2017
> 2. Big Sur 11.4
> 3. AIDente 1.04

Hi @eianzhang,

Does your MacBook still not charge above 95%?

Did you try an SMC reset on your MacBook? (You can find a detailed description on how to do an SMC reset on your MacBook here: https://support.apple.com/en-us/HT201295)

MatthiasKerbl, 2021-07-22:
> 1. MacBook Pro late 2013
> 2. MacOS 11.4
> 3. AlDente 2.2
>
> I don't see a Remove button in my AlDente settings. There is only a Reinstall Helper button.

Hi @Resserene,

Is your MacBooks battery still stuck at 1%?

If yes, could you explain in detail the exact sequence of events on how your MacBook got stuck at 1%? Since AlDente doesn’t even allow a charge limit that low.

eianzhang, 2021-07-23:
> Hi @eianzhang,
> 
> Does your MacBook still not charge above 95%?
> 
> Did you try an SMC reset on your MacBook? (You can find a detailed description on how to do an SMC reset on your MacBook here: https://support.apple.com/en-us/HT201295)

It still cannot be charged above 95%, sometimes it can reach 95.1%or 95.2%, then it discharges back to 95%. (in the process of long time charges from under 50%)

I have tried it twice,  SMC reset didn't work

MatthiasKerbl, 2021-07-23:
> > Hi @eianzhang,
> > Does your MacBook still not charge above 95%?
> > Did you try an SMC reset on your MacBook? (You can find a detailed description on how to do an SMC reset on your MacBook here: https://support.apple.com/en-us/HT201295)
> 
> Hi @MatthiasKerbl
> 
> It still cannot be charged above 95%, sometimes it can reach 95.1%or 95.2%, then it discharges back to 95%. (in the process of long time charges from under 50%)
> 
> I have tried it twice, SMC reset didn't work

Which program do you use to read out the battery percentage? Since macOS will never charge your MacBook to 100%.

eianzhang, 2021-07-24:
> Which program do you use to read out the battery percentage? Since macOS will never charge your MacBook to 100%.

[coconutbattery](https://www.coconut-flavour.com/coconutbattery/)

Before using AlDente, the battery can be charged to an interval between 96% - 99%, the max percentage varies each time.

It now stops charging accurately at 95% every time.

Resserene, 2021-07-24:
> Hi @Resserene,
>
> Is your MacBooks battery still stuck at 1%?
>
> If yes, could you explain in detail the exact sequence of events on how your MacBook got stuck at 1%? Since AlDente doesn’t even allow a [...]

Hi,
Yes, still stuck at 1%. Earlier it was stuck at 94% (I had set it at 90%). I reset the SMC multiple times but nothing improved. Even though it was stuck at 94% the real charge in the battery seemed to be 0 because my laptop would immediately die if I removed the power supply. Then a few days later it showed 1% on boot and since has been stuck at that. 

I was considering getting the battery replaced but I'm not sure if AlDente would let me even charge it since it's not letting it go beyond 1%. Finally had to give up and buy a new Mac mini because there doesn't seem to be a solution for this. You guys should really look into this. The default rate may not be high but for those that are suffering from the bug are really getting screwed.

MatthiasKerbl, 2021-07-29:
> [coconutbattery](https://www.coconut-flavour.com/coconutbattery/)
>
> Before using AlDente, the battery can be charged to an interval between 96% - 99%, the max percentage varies each time.
>
> It now [...]

Do you have "Manage battery longevity" in macOS under System Preferences/Battery/Battery/Battery Health... activated or deactivated or did you change this setting in the past?

Here is what I found:
I had "Manage battery longevity" deactivated for a year now and my MacBook(15" 2016) would always charge to 100% on the "real battery scale". A couple of days ago, I activated "Manage battery longevity" again because of curiosity. After a couple of cycles/days, the result was that the Battery would always stop charging at 95%, sometimes 95.2% but never more than that. Afterward, I deactivated "Manage battery longevity" again and after a couple of cycles/days and an SMC reset, my MacBook gets charged to 100% on the "real battery scale".

In conclusion, "Manage battery longevity" does exactly what it should "As your battery ages, peak capacity is reduced to extend battery lifespan." Unfortunately, "Manage battery longevity" changes take a long time to be activated/deactivated (days/many cycles and even an SMC reset).

Lastly, AlDente does not change any of these settings and it did not interfere or cause this behavior for me. My guess is that an SMC reset, macOS update, or manual change of this setting resulted in your max capacity being limited.

eianzhang, 2021-07-31:
> Do you have "Manage battery longevity" in macOS under System Preferences/Battery/Battery/Battery Health... activated or deactivated or did you change this setting in the past?
>
> Here is what I found: [...]

yeah, the second time I opened AIDente, its notification of turning off Optimized Battery Charging  made me turn both the
Optimized Battery Charging and Battery Longevity. Now I can figure out what happened, thanks a lot for your instruction.