AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #332: Keeps charging even when active. Updated to Monterey beta 3.

dmelgar, 2021-07-20:
Laptop awake, I'm using it. AlDente 2.2 installed. I reinstalled the helper. Rebooted. Charge limit set to 75%. It kept charging right through it. At 89% now when I noticed. This used to work with beta 1. Not sure with beta 2.

MatthiasKerbl, 2021-07-20:
Hi @dmelgar,

AlDente works fine for us on Monterey beta 3.

You should try to uninstall AlDente Classic 2.2 and install the newest version AlDente 1.04
Which MacBook model do you use?

dmelgar, 2021-07-20:
FYI: I attempted to remove AlDente Classic 2.2. I did the following steps
- Using Aldente, I set charge to 100%
- I scanned the filesystem for the helper, `find . -name com.davidwernhart.Helper -print 2> /dev/null`
- I removed all instances found.
- I shutdown the laptop for several minutes
- Started laptop, installed Aldente 2.2 again. It seems to be working. The settings page seems different. The Intel option no longer appears. Not sure why it appeared before and not now.
- When I first started it said, helper not found. I "reinstall helper", then rebooted. Shows up. 
Bottom line it appears to be working now.

This is on an M1 MacBook Air.
How is v 1.0.4 newer than 2.2? Is that a typo? Is 1.0.4 really the newer version?

dmelgar, 2021-07-20:
Looks like 1.0.4 is for a different product and is not the source code in this project nor is it the source code listed as being the source for that release.

That is not cool. There is no longer relation between that release and this repo. Its fine if you want to create your own product that is closed source, but don't list is in this repo as if its source resides here. It does not. 

Makes it less trustworthy. Can't tell what it may contain.

The source associated with release 1.0.4 is the old 1.0.4 from this repo and is NOT the source used to build the 1.0.4 release.

Why the confusion and deception?

MatthiasKerbl, 2021-07-20:
I think you launched AlDente with Rosetta which resulted in AlDente not working properly.
However, it should run natively now because the Intel Mode is no longer available.

About the versioning, take a look at this answer: #329

MatthiasKerbl, 2021-07-21:
> Looks like 1.0.4 is for a different product and is not the source code in this project nor is it the source code listed as being the source for that release.
> 
> That is not cool. There is no longer a relationship between that release and this repo. Its fine if you want to create your own product that is closed source, but don't list it in this repo as if its source resides here. It does not.
> 
> Makes it less trustworthy. Can't tell what it may contain.
> 
> The source associated with release 1.0.4 is the 2.0 alpha (as described in the code) from this repo and is NOT the source used to build the 1.0.4 release.
> 
> Why the confusion and deception?

Hi @dmelgar,

We are sorry if we broke some kind of Github rule by publishing our new versions of AlDente as a closed source app. It was never our intention to mislead or deceive anyone by using the newest versions of AlDente. Thank you for pointing out this mistake of ours. I have updated our readme to make it clearer for everyone when the change from open source to closed source happened.

If you have any other suggestions on improving our communication and transparency, let us know.

Kind regards,
Matthias

closed 2021-07-21 by MatthiasKerbl