AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #333: Does not work on macOS Monterey

lblomenkemper, 2021-07-22:
Hi,

I'm using AlDente Pro 1.04 and it doesn't seem to work with macOS 12 beta.

It seems like AlDente just randomly allows/disallows charging. I can't see any pattern, none of the modes/buttons have any effect.

I have already reinstalled AlDente and reset NVRAM/PRAM, unfortunately without success.

[image]
[image]

```
***AlDente Pro Debug File***
Main App Version: 1.04 Helper Version: 5
Model: MacBookPro16,1
Timestamp: 2021-07-22-12:18:06
Current State: charging
SMC KEYS:
isActivated: true
isInitialized: true
appleSilicon: false
thunderbolt: true
isCharging: true
isDischarging: false
chargerConnected: true
isClamshell: false
batteryTemperature: 36.597656
KEY: BCLM VALUE: 100 EXISTS: true
KEY: ACIN VALUE: 0 EXISTS: false
KEY: AC-W VALUE: 3 EXISTS: true
KEY: TB0T VALUE: 0 EXISTS: false
KEY: CH0C VALUE: 0 EXISTS: false
KEY: CH0B VALUE: 0 EXISTS: true
KEY: MSLD VALUE: 0 EXISTS: true
KEY: BRSC VALUE: 0 EXISTS: false
KEY: CH0I VALUE: 0 EXISTS: false
```

davidwernhart, 2021-07-22:
Hey @lblomenkemper 

Sorry to hear that something is not working right for you.
I am currently testing AlDente on my 16' MacBook Pro with the newest Monetary Beta and everything works without any problems for me.
Also, everything looks fine on the debug info you included, the SMC keys are set up in a way that should allow charging on your machine.
Could it be that you have optimised battery charging enabled? Maybe it automatically enabled again for you with the update to macOS 12?
Other than that, the only other thing that would come to my mind is that your battery might be miscalibrated in your case? As mundane as it might sound, have you restarted your computer since you first encountered that problem?

Best regards,

David

lblomenkemper, 2021-07-22:
Hi @davidwernhart

Thank you for the quick response!

Optimized battery charging is not enabled and I rebooted a couple of times since this problem first occurred, but it has occurred again and again with AlDente installed (I uninstalled it for about a week and had no problems during that time).

I will monitor this a bit more closely in the future and see if I can see any regularity, if so I will report on it here.

I just rebooted and my MacBook now reports that it is fully charged, so your assumption was correct.

closed 2021-07-22 by lblomenkemper