AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #335: "sailing mode" general explanation - questions

RalphKingsman, 2021-07-23:
I Would greatly appreciate your thoughts on the following two questions:

1. On apphousekitchen I see a "sailing mode" and a special page with a feature explanation. It's a good read and I understand it completely.
Your statement "Since it is theoretically healthier for the battery to be charged 10% once instead of 1% 10 times": however logical that sounds to me, is there a source / scientific test for this?

2. "While the charging of the MacBook is paused, i.e. the charge level is exactly at the set charge limit, the power supply is primarily used as a power source."
Is this true? It is my understanding that _all power_ needed to operate today's Macbooks is drawn _directly from the battery_ and that the battery - in turn - is charged to top up the battery, i.e. there is no way to draw power directly from the charger.
This is contrary to vintage Macbooks (pre 2010) with replaceable batteries where the unit could operate even without the battery inserted.

As the "sailing mode" would be the only reason for me to buy the Pro version, I'm interested to hear what your thoughts are on this.

Best,
Ralph Kingsman

MatthiasKerbl, 2021-07-23:
> I Would greatly appreciate your thoughts on the following two questions:
>
> 1. On apphousekitchen I see a "sailing mode" and a special page with a feature explanation. It's a good read and I [...]

Hi @RalphKingsman,

1. The Sailing Mode was a highly requested feature from multiple people in our userbase. For this reason and with the theoretical benefits, we decided to implement this feature. However, I have not yet seen a specific study on this topic since it's kind of a niche use case. If you find any research in this regard, let me know.

2. I am not yet 100% sure how everything regarding this topic works and I am still learning about this. However, here is what I have learned so far. Firstly, when you set a charge limit with AlDente and the charge limit has been reached, the charging/discharging wattage and the battery amperage drops to 0 W. Even though, the power drawn from the power brick and the power used from the MacBook is approximately the same and fluctuates between 15W and 65W in my case. Secondly, when clicking on the macOS battery icon while the charge limit has been reached, macOS says "Power Source: Power Adapter" and "Battery is not charging". Therefore I think that even though you can't run newer MacBooks without a battery, it is possible to "bypass" the battery and use the power brick directly as the primary power source. However, as mentioned before, I still have to look into this topic in more detail.

Kind regards,
Matthias

MatthiasKerbl, 2021-08-02:
Hi @RalphKingsman,

I will close this issue for now since I have answered your questions.
If you have more questions, feel free to ask.

closed 2021-08-02 by MatthiasKerbl