AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #338: Issue with macOS 12.0 (21A5284e)

Samuels1950, 2021-07-23:
For me, the Al Dente Pro loads but no longer limits the charge since I installed this beta. Nothing else has changed on my M1 MacBook Pro. Worked fine with the prior Monterey betas, Thoughts?

MatthiasKerbl, 2021-07-23:
> For me, Al Dente Pro loads but no longer limits the charge since I installed this beta. Nothing else has changed on my M1 MacBook Pro. Worked fine with the prior Monterey betas, Thoughts?

Hi @Samuels1950,

Have you tried a complete reinstall of AlDente Pro?

Samuels1950, 2021-07-24:
Yes. I uninstalled Pro. But, when I reinstall using the link in my order confirmation email I keep getter the initial beta version and that really doesn't like Monterey. How do I download the current version of Pro? I can't seem to fins a link.

MatthiasKerbl, 2021-07-25:
> Yes. I uninstalled Pro. But, when I reinstall using the link in my order confirmation email I keep getter the initial beta version and that really doesn't like Monterey. How do I download the current version of Pro? I can't seem to fins a link.

You can download the latest version here: https://github.com/davidwernhart/AlDente/releases/download/1.04/AlDente_1.04.dmg

Samuels1950, 2021-07-26:
I've successfully reinstalled but the app is ignoring the charging limit that I set. All the settings are correct and optimize battery is turned off in settings. This just started with the current beta. Perhaps it will be better when the next beta drops. I am not suggesting that beta os should be supported and I know things will be right when Monterey is publicallt released.

MatthiasKerbl, 2021-07-27:
> I've successfully reinstalled but the app is ignoring the charging limit that I set. All the settings are correct and optimize battery is turned off in settings. This just started with the current [...]

Hi @Samuels1950,
Please try these steps suggested by @dev-song #339

n-hass, 2021-09-28:
Wanted to add I am also experiencing this issue on both AlDente Classic 2.2 and the 'new' AlDente Free 1.07. On launching the app, can set charge limit and is respected until Mac sleeps at least once, sometimes requires more sleep sessions (can't identify pattern for duration etc., seems random / intermittent).

Re-launching the app works as a temporary fix to reinstate the charge limit and allow discharging, but will fault again as detailed above. Process written in #339 provides the same temporary fix.

**System**
Device: MacBook Pro (16-inch, 2019)
Processor: 2.4 GHz 8-core Intel Core i9
macOS version: Monterey, 12.0 Public Beta (build 21A5522h)