AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #339: Issue with Aldente not working after MacOS update

dev-song, 2021-07-27:
This issue is to share my solution to solve the problem that Al Dente does not work properly after MacOS software update.
I hope this solution can be used commonly to help others with the same problem.

### Common behavior after MacOS update
AlDente's `Max. Battery Charge` is reset to default value, 103 (Which I previously set to 70)
Re-lowering this number to other value doesn't work, as the device doesn't stop charging even after restarting it.
[README.md](https://github.com/davidwernhart/AlDente) tells that reinstalling AlDente might solve the problem.

### My solution for the problem
1. Unplug MacBook's power cable
2. Restart it
3. Wait 'til device is fully initialized (didn't test the case without waiting)
4. Re-plug the power cable
5. Check if MacBook stops charging

### Checked last 3 updates
- 11.3 -> 11.3.1
- 11.3.1 -> 11.4
- 11.4 -> 11.5.1

jakuSk, 2021-12-12:
Hi, Unfortunatelly nothing worked for me. Full uninstall that was done step by step without any visible changes, then @dev-song solution without any luck.
OS Monterey 12.0.1 (fresh install)
Al dente 1.11/1.10

Discharge works, then limiter does not. Macbook charges to etc 50%, if the limiter is set over 50% say 55% then baterry will stop charing at that level.

MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports) 65W charger / 30W original Apple charger without any differences.

jakuSk, 2021-12-19:
Monterey 12.1 did not help

MatthiasKerbl, 2022-11-07:
> Hi, Unfortunatelly nothing worked for me. Full uninstall that was done step by step without any visible changes, then @dev-song solution without any luck. OS Monterey 12.0.1 (fresh install) Al dente [...]

Hi @jakuSk,

I am sorry you are experiencing this issue.
It is a known issue for some MacBook models and you can find more information about this issue here #515

Cheers,
Matthias

MatthiasKerbl, 2022-11-07:
Hi @dev-song,

We have a new thread about this issue with the newest updates here #768
Therefore, I will close this issue but feel free to add any new findings to the issue mentioned.

Cheers,
Matthias

closed 2022-11-07 by MatthiasKerbl