AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #340: Discharge is deactivated often

SenchoPoro42, 2021-07-28:
Hello, I upgraded from the fully functional AlDente 2.2 to AlDente Free 1.04. I like the design of the new application but the discharge seems to not function as intended?

If I have restarted my computer, or let it sleep overnight (most often) without the power cable in both cases, it will be below my set battery charge of course. I've set it to 80%. AlDente is running on boot.
If I then charge my machine like normal, it will go above the set charge to 100% every time.

I have to manually click the Discharge button in the application before it will stop charging and then drain the battery to 80%. At this point when it hits my battery threshold it will function like expected where it will use power from the cable and not charge the battery.

Is this intended the intended process of the new AlDente Free version?

MatthiasKerbl, 2021-07-28:
Hi @SenchoPoro42,

Normally, AlDente should always limit charging at the set charge limit when your MacBook is turned on.

Which MacBook model do you have?
Which macOS version do you use?
Did you completely uninstall and delete AlDente Classic 2.2?

SenchoPoro42, 2021-07-28:
It's a Macbook Pro M1 2020
Currently MacOS Monterey Beta 21A5284e
I took AlDente Classic and moved it into Trash and emptied the bin,  please advise on other ways of removing software on Mac.

I just restarted it at 82% and took a screenshot of it having disabled the Discharge button on boot.
[image]

MatthiasKerbl, 2021-07-28:
> It's a Macbook Pro M1 2020
> Currently MacOS Monterey Beta 21A5284e
> I took AlDente Classic and moved it into Trash and emptied the bin, please advise on other ways of removing software on Mac.
>
> I just [...]

It looks like the Charge Limiter works fine.
In regards to the Discharge feature, its functionality has changed a little bit since AlDente Classic 2.2. Due to too many bugs, Discharge has to be activated manually when your desired charge limit is lower than your current SoC.
You can read more about the details of the feature in our blog: https://apphousekitchen.com/feature-explanation-discharge/

closed 2021-08-25 by MatthiasKerbl