AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #342: Limiting the charge level during Top-Up

SYDT000, 2021-08-01:
I am a pro subscriber of AlDente. Sometimes I use the top-up to increase the charge level while the battery temperature remains high, but I want to charge it to a limit- 80%. I set it at 80 since the default limit goes high at 100% for top-up, but it keeps charging over 80%. I think it is easily fixable.

MatthiasKerbl, 2021-08-02:
Hi @SYDT000,

Please answer these questions first:

1. Which MacBook model do you use?
2. Which macOS version do you use?
3. Which AlDente Pro version do you use?

I am not sure if I understood your question correctly. Are you saying that the Charge Limit feature does not work for you?

General things about the Top Up feature: When Top Up is activated, your MacBook will always charge to 100% even if the temperature is above the Heat Protection Temperature. If you want to charge your MacBook to 80% you have to use the Charge Limiter feature for that. If your MacBook stops charging to 80% because the temperature is above the Heat Protection temperature but you want it to charge to 80% anyways, you have to deactivate the Heat Protection feature. Furthermore, it is not possible right now to change the charge limit of Top Up from 100% to something else.

SYDT000, 2021-08-03:
Hi,

Macbook Model A1706 
- MacOS 11.5.1
- Aldente 1.04

So you have the option to slide the charge level while topping up. If topping up forces the system charge way up to 100, what is the point of having the slider move up or down? You could have just disabled it to avoid confusion. Even better, if we could charge it or top-up up to a certain level. Hope it helps.

SYDT000, 2021-08-03:
One correction, actually when I change the slider position, it changes the charging mode and stops charging. But If I manually enter the charge percentage such as 80, it keeps me on the top-up mode while the slider indicates 80 percent limit but the top-up brings the charging level above 80 towards 100. Looks like a bug?

MatthiasKerbl, 2021-08-04:
> Hi,
>
> Macbook Model A1706
>
> * MacOS 11.5.1
> * Aldente 1.04
>
> So you have the option to slide the charge level while topping up. If topping up forces the system charge way up to 100, what is the point of [...]

Hi,
Thank you for the suggestions. We are thinking about extending the functionality of the Top Up feature in the future.

MatthiasKerbl, 2021-08-04:
> One correction, actually when I change the slider position, it changes the charging mode and stops charging. But If I manually enter the charge percentage such as 80, it keeps me on the top-up mode [...]

Yeah, that looks like a bug. Will be fixed with the next update.

closed 2021-11-01 by MatthiasKerbl