AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #347: Macbook Air M1 turning off at 20% Battery

Rixxx-git, 2021-08-07:
Hi guys,
did anybody of you also experience that the Macbook turns off, although it shows about 20% battery remaining? When I plug it in after that and it reboots it shows about 10% battery. 

My battery stats say: 
- 44 cycles (I have the macbook plugged in all day while working with it about 8h per day and al dente set to 60%)
- battery health: normal
- max. capacity: 100%

Am I using Al Dente wrong? Or is this not because I use Al Dente the way I do?

Thanks for your help!

msurma, 2021-08-07:
Do a full recalibration of your battery, as described here: https://apphousekitchen.com/feature-explanation-calibration-mode-2/

Rixxx-git, 2021-08-07:
Thanks so much for your quick and helpful reply!

closed 2021-08-07 by Rixxx-git