AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #348: AlDente Phoning Home And Privacy?

mikeatlarge, 2021-08-09:
My firewall indicates even the free version of AlDente contacts multiple servers which could be a privacy concern. Perhaps the developer can explain what's going on and if internet access is required for the free and pro version (once the license is verified) to function? I'm reluctant to pay for something that runs all the time with potential spyware built-in. Thanks in advance.

davidwernhart, 2021-08-09:
Hi @mikeatlarge 

AlDente does connect to the internet because of two reasons:

- To check for updates periodically. It simply polls an XML file on the AppHouseKitchen server and prompts you with the update message if it finds that there is a new version.
- To check if a possible license is valid using Paddle's server (our payment provider). This is done on both versions since the free and the pro version are the same binary now.

In any case, the app does not track any user data or other statistics.

For the free version, nothing should change when disconnected from the internet besides not receiving potential updates.
The pro version verifies the the license on every start, but pro features are only disabled when there have been 10 days without a successful license verification. Other than that, it can also be used offline to the full extend.

Hope that clears things up!

Best regards,
David

mikeatlarge, 2021-08-09:
David, thanks so much for the prompt reply! That seems reasonable enough. I'll play around with the free version on both my M1 Air and older MacBooks and likely buy a lifetime license if it works well on all 3. I'm honestly surprised there are not more apps like this. There are a couple apps that send a notification and/or ring a chime when the battery charges to a certain percentage but they don't cut off the charging. And the Big Sur native adaptive algorithm has been just about worthless for me. It nearly always charges to 100% no matter what I do and it's not even available on older MacBooks running Big Sur. Thanks again.

closed 2021-08-09 by mikeatlarge