AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #351: High CPU load

wazar, 2021-08-13:
Using Pro (1.06)
Macbook Pro (M1, 2020)
Latest macOS Big Sur

AlDente uses 99.99% CPU in activity monitor and sometimes hangs using 800% (lots attached)
[aldentepro-debug-2021-08-13-18:27:42.pasta.zip](https://github.com/davidwernhart/AlDente/files/6985321/aldentepro-debug-2021-08-13-18.27.42.pasta.zip)
[image]

JongYun-Kim, 2021-08-15:
I've got the same issue...

100% cpu used.
750-800% cpu is used when I adjust the charge limit...

This makes my MacBook so hot. It's never been this hot before...

I uninstall the free version and installed this pro version on SetApp. 
M1 MBA (11.4 and 11.5.2)

[image]

Rag0n, 2021-08-16:
Same issue, I'm using AlDente Pro from Setapp and I have MacBook Pro m1.

J0n4t4n, 2021-08-16:
Same, MBA M1 through Setapp.

JongYun-Kim, 2021-08-16:
I heard from AppHouseKitchen that they are working on it. They can't reproduce the bug at the moment though..
And, They added that "The one on SetApp is actually the newest version, which we have not released on GitHub yet. However, we will do so soon."

davidwernhart, 2021-08-18:
Hi Guys!

We are aware of this bug and have finally been able to replicate it.
A fix will be released soon!

Thank you for your patience,

David
AppHouseKitchen

JongYun-Kim, 2021-08-18:
> Hi Guys!
> 
> We are aware of this bug and have finally been able to replicate it.
> A fix will be released soon!
> 
> Thank you for your patience,
> 
> David
> AppHouseKitchen

Thank you very much.

Now that I hear that the new version is released soon, I can't be happier!

davidwernhart, 2021-08-19:
We have just released our new Version 1.07 which fixes this bug. Sorry for the inconveniences and kind regards, David

closed 2021-08-19 by davidwernhart