AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #356: [Feature request] Sailing mode - v1.03 style

Loeivy, 2021-08-20:
Since Sailing Mode seems to have changed in 1.04 and beyond, I'd like to request a standalone feature to have sailing mode as it was in 1.03 and before.
Since 1.04 it seems to not be possible to keep sailing after unplugging the MacBook from it's power supply, in 1.03 that was possible.

To explain the problem in a little more detail:
I often use my MacBook with an external monitor and a "one cable solution" - this means that, whenever I use the monitor, my MacBook is going back into charging. The fact that sailing mode doesn't prohibit charging in this case makes it a little bit useless since my MacBook will also charge back to the set amount if it only lost 1-2%.

In 1.03 however, setting the charge limit to 80% for example, and sailing to 30% would result in the MacBook not charging back up to 80% whenever it's plugged back in. So, example given I use my MacBook on Battery and it's at 62%, I plug it back in and it will stay at 62%. Next time I use it it drops to 37% and it will charge back up to 80%. Like this, the battery will not constantly topple back up to 80% but use a wider range of it's capacity instead, making the whole feature of sailing much more useful to use cases like mine and likely many others (as many seem to mention that self-discharge is relatively slow/irrelevant anyway)

Besides, MacBooks have an integrated sailing mode for the first 7 or so percent anyway. It's also why the top 6% shown in CoconutBattery for example differ from the % macOS shows. The actual 100% range of MacBook batteries are 94% in truth (same for iPhones) - this is also why the first drop from 100% to 99% seems to take much longer than any other drop. So, all in all, Sailing mode is kind of pointless as it is right now and even lacks behind integrated features of macOS in case of 100% charge.

MatthiasKerbl, 2021-08-22:
Hi @Loeivy,

Thank you for your suggestions. We are working on updating the Sailing Mode to do most of the things you suggested.

Kind regards,
Matthias

Loeivy, 2021-08-23:
Thank you for the reply @MatthiasKerbl,

is there any estimated date for this feature to be implemented in the release? Or maybe any kind of beta version you need help testing with?
Also, if you're interested, I'd like to offer localising the application to German, if there are any plans to implement localisations for AlDente in the future. (I couldn't find any localisation related files in the repository so far, so maybe there might be any plans for this in the future?)

Kind regards

MatthiasKerbl, 2021-08-25:
> Thank you for the reply @MatthiasKerbl,
>
> is there any estimated date for this feature to be implemented in the release? Or maybe any kind of beta version you need help testing with?
> Also, if you're [...]

Right now, I can’t give you a timeline for the updated Sailing Mode.
Thank you for your offer of translating the AlDente to german. We have started working on implementing localisations and we will come back to you when it is ready.

dalisoft, 2021-09-01:
For me same issue. Sail mode resets between charger unplug-plug

yyj603, 2021-09-11:
same issue here ,would like to have it back, thanks

marcogracklauer, 2021-10-20:
Any Timeline for this feature?