AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #357: Macbook cant sleep

SiposPtr, 2021-08-20:
The app itself works great,and i was happy with it.
But it turned out that it prevents my macbook from staying in sleep mode.
As soon as i put it in sleep mode, it wakes up on itself.
Ive reinstalled my mac,and then everything was fine.
Then decided to give AlDente a second shot, and did the same as the first time,it prevented my mac from staying in sleep mode:(

MatthiasKerbl, 2021-08-25:
Hi @SiposPtr,

Which MacBook model do you use?
Which macOS version do you use?
Which version of AlDente do you use?

Could you elaborate on your problem a little bit more? When exactly does AlDente prevent your MacBook from sleeping? When it is charging to the charging limit, when charging is paused at the charging limit,...?

SiposPtr, 2021-08-25:
hi,
MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)
macOs Big Sur 11.5.2
AlDente 1.0.7

Ive set the charging limit to 80%, and it  does keep it at that percent, so thats not a problem.
It pauses at the charging limit, so basically the app works perfectly.

But once i tried to lock my mac with either the Sleep button from the  menu from the Menu bar, or with the Option+Command+Power combinations, it didnt stay in sleep mode, it woke up immediately.
Even if i did lock it, woke up, and then locked it again, it always woke up.

Unfortunately i dont have the logs, but it did happen again, with a fresh installed macOS.

MatthiasKerbl, 2021-08-25:
Unfortunately, I am currently not able to reproduce your bug on my Intel-based MacBook.
Do you use your MacBook in clamshell mode?
Did you try to completely uninstall AlDente, restart your MacBook, and install AlDente again?

SiposPtr, 2021-08-25:
Im not using my mac in clamshell mode. Could this be the only problem?

dalisoft, 2021-09-01:
In calibration mode **sleep mode** does **not work**. Please check to make sure you're not in **calibration** mode

MatthiasKerbl, 2021-11-15:
Hi @SiposPtr,

We have just released a new version of AlDente (1.10). https://github.com/davidwernhart/AlDente/releases
This version should fix the issue you encountered.

Cheers,
Matthias

closed 2021-11-15 by MatthiasKerbl