AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #359: Pro Version doesn't prevent full load of battery after the computer was shut down

TGHein, 2021-08-25:
I recently purchased Al Dente Pro but I have troubles that I never had using the free version of Al Dente. I am running it on a MacBook Pro 2019 with MacOS 11.5.2 installed. The problem is that the battery will always be charged completely (=100%) whenever the computer is completely shut down and left connected to the charger. Even if I had set the max. charge to e. g. 60% it fully loads the battery up to 100% when it is shut down.
Please help!

MatthiasKerbl, 2021-08-25:
Hi @TGHein,

Thank you for purchasing AlDente Pro.
Do you have the setting "Stop charging when powered off" enabled? (You can find this setting in Settings/Charge Settings)

TGHein, 2021-08-25:
Hi Matthias,
thank you for getting back to me. The setting (of Al Dent Pro) "Stop charging when powered off" is currently disabled. Do I need to enable it?

MatthiasKerbl, 2021-08-25:
Yes, if you want to pause charging while your MacBook is turned off, you have to enable "Stop charging when powered off". However, keep in mind that AlDente can’t control the charging behavior of your MacBook while it is turned off. Therefore, it will just pause charging at the current percentage and not charge to the charge limit.

closed 2021-08-25 by MatthiasKerbl

TGHein, 2021-08-25:
Thank you for your quick answer. I wonder though why the standard version of Al Dente (not the pro version) always kept the charge limit set not matter if the MacBook was turned on or off. Was this developed on a different basis?

MatthiasKerbl, 2021-08-25:
Yes, this was developed on a different technical pathway. If you prefer the old way, you can activate the Intel Mode in settings and you will get the old functionality. However, most of the Pro features don’t work anymore, when the Intel Mode is activated.