AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #360: v.1.07 charges for about one minute after connect charging

brisk84, 2021-08-25:
I was start using AlDente Classic 2.2 with MacBook Air M1. All was fine, but two days ago I'm updated to v.1.07 and I see that it is not stop charging at the same time when charger is connected. It takes about one minute while charging is activated and then it stops. Version 1.04 stops charging immediately, and AlDente Classic 2.2 too.
Now I'm switched to v.1.04, but think what if this is feature of v.1.07?
I was trying to uninstall and reinstall v.1.04 and 1.07 several times, and it was always the same result.

MatthiasKerbl, 2021-08-25:
Hi @brisk84,

This is definitely not a feature of AlDente 1.07 and I do not encounter this behavior of AlDente.
Did you completely uninstall all older versions of AlDente and restart your MacBook before installing AlDente 1.07?

brisk84, 2021-08-25:
@MatthiasKerbl, thank you for response!
I try to uninstall, reboot, install again and nothing changes, but then I remove /Applications/AlDente.app, reboot and install v.1.07 and it works fine.
Thank you again, you make a great app!

closed 2021-08-25 by brisk84