AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #365: Calibration Mode Warning

simotau, 2021-08-28:
Hi everyone!

When initiating Calibration Mode, I get this warning: "To enable discharge in all scenarios, like clamshell mode, AlDente has to temporarily disable your device's sleep functionality. Because of this, the Sleep option might be greyed out while discharging on some MacBooks. Also, some other apps controlling sleep behaviour might not work correctly."

What does it mean? If I use the Calibration Mode or the Discharge Mode I will not be able to put my Mac to sleep anymore, until I disable the options?

Thank you for your attention and keep up with the great work!

Simone
MacBook Pro M1 8/256
Aldente Pro v. 1.07

MatthiasKerbl, 2021-08-28:
Hi Simone,

> What does it mean? If I use the Calibration Mode or the Discharge Mode I will not be able to put my Mac to sleep anymore, until I disable the options?

Yes, AlDente has to disable sleep for Discharge and Calibration Mode to work during clamshell mode.

Thank you!

Cheers,
Matthias

closed 2021-08-28 by MatthiasKerbl

simotau, 2021-08-29:
Thank you for your reply!

I've gotta say that I preferred the way Calibration Mode worked in the older version, but hey! One can't have everything in this world XD

Thanks again!

Simone